realsenseai/librealsense #11126: L515-gyro and accel data synchronization

Ceciliammm, 2022-11-20:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1371405|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003cbb76779|
|**Firmware Update Id**|0003cbb76779|

I want to export synchronized gyro and accel data. At the same time, obtain the corresponding timestamp of the exported gyro and accel data. How can I realize it?
In addition, the language that I use is Python.
Thank you in advance.

MartyG-RealSense, 2022-11-20:
Hi @Ceciliammm  If you wish to export the gyro and accel data then you could record these streams to a bag file with the **enable_record_to_file** SDK instruction using Python code such as the simple example below.

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 100)
config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100)
config.enable_record_to_file("BagFile.bag")
auto dev = pipe.start(config).get_device().as<rs2::record>();
```

Simultaneously reading timestamps from a bag file that was still being recorded could be problematic though.  It may be best to display the timestamps live - like in the Python script at https://github.com/IntelRealSense/librealsense/issues/5291 - as the IMU data is being recorded into the bag.

MartyG-RealSense, 2022-11-27:
Hi @Ceciliammm  Do you require further assistance with this case, please?  Thanksl

closed 2022-12-04 by MartyG-RealSense