realsenseai/librealsense #11127: Troubles compiling a ROS project because of my realsense library

a-laborie, 2022-11-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 5.13| 
| Operating System & Version |  Linux (Ubuntu 20.04) /   | 
| Kernel Version (Linux Only)    |  (5.15.0-53-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |  { 2}                          | 
| Language                            |  {python/c++ }                          | 
| Segment			|  {Robot/ }                          | 

### Issue Description
Hi everyone, 
I try to use both the  camera and the  robotic arm. I now can control the camera thanks to the  library that I installed from this source. I want to use  to control my robotic arm in python. Though, it seems that my  library is wrongly installed, because every time I try to compile my project (`catkin_make`), it fails with the following message:
```
-- +++ processing catkin package: 'realsense_camera'
-- ==> add_subdirectory(realsense/realsense_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /home/alaborie/catkin_ws/devel/share/librealsense/cmake/librealsenseConfig.cmake:173 (message):
  Project 'realsense_camera' tried to find library 'realsense'.  The library
  is neither a target nor built/installed properly.  Did you compile project
  'librealsense'? Did you find_package() it before the subdirectory
  containing its code is included?
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  realsense/realsense_camera/CMakeLists.txt:39 (find_package)
```
Could you help me? I am new in ROS so I am quite confused by this error. I don't understand what is wrong with the installation of the _realsense_ library.

MartyG-RealSense, 2022-11-21:
Hi @a-laborie   **realsense_camera** was an old ROS wrapper for RealSense that has not been updated for years and does not support modern RealSense camera models such as D435i as it was designed for use with early models such as SR300, F200 and R200.  The current **realsense2_camera** wrapper that supports the 400 Series models can be found at the links below.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

**ROS2**
https://github.com/IntelRealSense/realsense-ros

MartyG-RealSense, 2022-11-27:
Hi @a-laborie  Do you require further assistance with this case, please?  Thanksl

a-laborie, 2022-11-27:
Hi, 
Thank you for your help. I think that my problem was that I had installed both old and recent libraries about the realsense camera which may be incompatible. I had to reinstall ROS and then I did as described on the ROS1 website you sent me. It now works !
Thanks again and best regards
AL

closed 2022-11-27 by a-laborie

MartyG-RealSense, 2022-11-27:
That's great to hear that you achieved a solution, @a-laborie - thanks very much for the update!