realsenseai/librealsense #11130: Number of Frames are very very less after aligning depth frames to rgb frames.

surajiitd, 2022-11-21:
--------------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | Open RealSense Viewer --> v2.51.1 | 
| Operating System & Version |    Ubuntu 20.04.5 LTS  | 
| Kernel Version (Linux Only)    |  5.15.0-53 | 
| Platform                                 | PC |
| SDK Version                            |  legacy / 2.24.23                          | 
| Language                            |  python  | 
| Segment			|  others| 

### Issue Description
Please find my attached code that I am using to align the depth with rgb frames, But I am getting very very less frames as expected. Let's say for a video of 4,5 min in 30fps, I am getting only 20 frames. 
Please help what is going wrong here in my code??
@MartyG-RealSense 

**CODE:**
``` python
pipeline = rs.pipeline()
config = rs.config()

print("*"*20,"Doing for",bag_file,"*"*20)
config.enable_device_from_file(bag_file)

config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)

pipeline.start(config)
align = rs.align(rs.stream.color)

depth_scale = .001
ind = 0
last_timestamp = 0.

try:
    while(True):

        frameset = pipeline.wait_for_frames()
        
        current_timestamp = frameset.get_timestamp()
        if current_timestamp < last_timestamp:
            break
        last_timestamp = current_timestamp
        
        
        aligned_frames = align.process(frameset)
        aligned_depth_frame = aligned_frames.get_depth_frame() 
        color_frame = aligned_frames.get_color_frame()
        

        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            print("not aligned_depth_frame or not color_frame")
            continue

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        #To convert BGR to RGB
        color_image = color_image[:,:,::-1]

        cv2.imwrite(os.path.join(depth_dir,f"{ind:06d}.png"), depth_image)
        cv2.imwrite(os.path.join(rgb_dir,f"{ind:06d}.png"), color_image)
        ind+=1

finally:
    pipeline.stop()
```
this is a video of 4,5 minutes. and only 35 rgb,depth images are generated from above code.
[image]

MartyG-RealSense, 2022-11-22:
Playback of a bag file can be more stable if **set_real_time** is set to False.  

The bag can be prevented from looping by adding `repeat_playback=False` to the brackets of the cfg.enable_device_from_file instruction.

These two principles are demonstrated in the Python script at https://github.com/IntelRealSense/librealsense/issues/6340

surajiitd, 2022-11-22:
Thanks @MartyG-RealSense 
I set both to `False` ,those issues resolved. Now I got error `RuntimeError: Frame didn't arrived within 5000` in my laptop . **But** I could generate all the rgb,depth in another PC(which has a faster compute). 
Thanks. Attaching my code where I made the changes, maybe helpful to others.
``` python
config.enable_device_from_file(bag_file, repeat_playback=False) """repeat_playback=False to don't repeat multiple times from start to end[so do at once]"""
rgb_dir, depth_dir = make_all_dirs(bag_file)

config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.rgb8, 30)

profile = pipeline.start(config)
playback = profile.get_device().as_playback()
playback.set_real_time(False) """ set_real_time(False) is used to NOT skip frames(which happens by default)."""
last_pos = playback.get_position()
```

closed 2022-11-22 by surajiitd

MartyG-RealSense, 2022-11-23:
Thanks very much @surajiitd for the update and for sharing your solution with the RealSense community!