realsenseai/librealsense #11133: wrapper for free pascal /Delphi

Blueicaro, 2022-11-22:
Hello, do you have plans to make a wrapper for free pascal or Delphi

/BlueIcaro

MartyG-RealSense, 2022-11-22:
Hi @Blueicaro  There are not currently plans to create a wrapper for free Pascal or Delphi.  If a member of the RealSense community is able to create one though then they are welcome to submit it as a **Pull Request (PR)** to be considered for inclusion into the RealSense SDK.

https://github.com/IntelRealSense/librealsense/pulls

MartyG-RealSense, 2022-11-29:
Hi @Blueicaro  Bearing in mind the above comment, do you require further assistance with this case, please?  Thanks!

Blueicaro, 2022-11-29:
Hello @MartyG-RealSense. I was looking for a wrapper. And I didn't find any thing. I asked here in case there was any plan.
So I started a proyect to make the wrapper: https://github.com/Blueicaro/Pascal-Librealsense. 
Of course I 'll need help. For the moment I make some test and every thing is working. I don't have my own camera. In my  work there is a camera (D435), and some weekends I take to the home. Many features I'll can't check them. 
Me doing the wrapping as a hobby and for the community. 
Thanks

MartyG-RealSense, 2022-11-30:
Yes, there is not an existing Pascal wrapper for librealsense, so your wrapper project is very welcome.  Thanks so much for creating and testing it!

Blueicaro, 2022-12-05:
Hello, I finished the translation of C include files.
You can find a wrapper in my githut: https://github.com/Blueicaro/Pascal-Librealsense
It's the first interation. 
/BlueIcaro

MartyG-RealSense, 2022-12-10:
Hi @Blueicaro  Do you require further assistance with this case, please?  Thanks!

Blueicaro, 2022-12-12:
Hello @MartyG-RealSense , thanks for asking. For the moment I finish the translate. I started to translate the examples.
Of I'll will help. So if you want, you can keep open this issue, and I'll write here.
/BlueIcaro

MartyG-RealSense, 2022-12-12:
I have added an **Enhancement** label to this issue to signify that it should be kept open.  Please do continue to add further details in comments.   Thanks!

Blueicaro, 2022-12-19:
Hello MartyG, after make some demos. I working to make a new demo. I want to save a frame as jpg. I was looking this code;: https://dev.intelrealsense.com/docs/rs-save-to-disk
But I have a question. In this line: 
// Use the colorizer to get an rgb image for the depth stream
if (vf.is<rs2::depth_frame>()) vf = color_map(frame); 
The frame is "colorizer", what is this. The frame doesn't containt a RGB image?.
/BlueIcaro

MartyG-RealSense, 2022-12-19:
You are correct, an RGB image is not involved.  A colorizer instead **color shades** a depth pixel to a color that represents the depth distance.

Blueicaro, 2023-01-29:
Hello, I was taking a look to the sample rs-save-to-disk, as I said before. But the colorize is a C++ method. I was diving into source but I didn't see how can do the work. Can you explain me it. Or provide a C sample.
Thanks

MartyG-RealSense, 2023-01-30:
The colorizer color-shades the coordinates of a depth image with colors that represent the depth value (distance from the camera) of a particular coordinate.  If the colorizer is not used then the depth image should still be generated but will be monochrome instead of shaded with colors ranging from blue to white.

So in the rs-save-to-disk C++ example, lines 22 and 41 could be commented out to remove the colorizer-related functions from the program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp#L22
https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp#L41

closed 2026-10-25 by MartyG-RealSense