realsenseai/librealsense #11138: Post Processing before alignment will impact performance and have different mesh result

YanHernTan, 2022-11-23:
Hi, I was trying to get the mesh results from streaming but I realize that if I align the frames after post processing, the mesh result will be way different than what I get if I align the frames before post processing. The FPS is also greatly impacted, where alignment after post processing will have around 1FPS while alignment before post processing will have around 10FPS. 

I am trying to achieve the same result with RealSense Viewer but it seems the mesh generated will be different. Is there any other configuration I should change to achieve the same mesh result in RealSense Viewer?

    pipeline = initialize_device()
    frame_count = 0
    align = rs.align(rs.stream.color)
    # post process initialization
    decimate = rs.decimation_filter(3)
    threshold = rs.threshold_filter(0, 10)
    depth_to_disparity = rs.disparity_transform(True)
    spatial = rs.spatial_filter()
    temporal = rs.temporal_filter()
    disparity_to_depth = rs.disparity_transform(False)

    try:
        while True:
            dt0 = datetime.now()
            # Get frameset of color and depth
            frames = pipeline.wait_for_frames()

            #frames=align.processs(frames)

            #frame post processing
            frames = decimate.process(frames)
            frames = depth_to_disparity.process(frames)
            frames = spatial.process(frames)
            frames = temporal.process(frames)
            frames= disparity_to_depth.process(frames)
            frames = threshold.process(frames).as_frameset()

            frames = align.process(frames)

            path = ASSETS_PATH + "mesh.ply".format(idx)

            exporter = rs.save_to_ply(path)
            exporter.set_option(rs.save_to_ply.option_ply_mesh, True)
            exporter.set_option(rs.save_to_ply.option_ply_binary, True)
            exporter.set_option(rs.save_to_ply.option_ply_normals, False)
            exporter.process(frames)

            process_time = datetime.now() - dt0
            print("\rFPS: " + str(1 / process_time.total_seconds()), end='')
            frame_count += 1

MartyG-RealSense, 2022-11-23:
Hi @YanHernTan  The RealSense Viewer exports a ply from its 3D **pointcloud** mode.  A pointcloud is generated by converting 2D pixel coordinates to 3D real-world coordinates measured in meters.

Your script is not generating a 3D pointcloud, so the ply exported from it will not be the same as the pointcloud data contained in a ply exported by the Viewer.

The RealSense SDK's **opencv_pointcloud_viewer.py** Python example program at the link below demonstrates producing a pointcloud with the **pc.calculate** and **pc.map_to** instructions, which map color coordinates onto depth coordinates to create an RGB-textured pointcloud without using alignment.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

In regard to whether to perform alignment after post-processing: whilst applying the post-processing filters first is highly recommended, there have been rare cases where an application has performed better if alignment is done before filtering.  So you should choose which approach works best for your particular project.

You could test though whether akign after filtering has a better FPS if you comment out the **Spatial** filter, as it can take a long time to process whilst not resulting in a significant difference to the image.

YanHernTan, 2022-11-24:
RealSense Viewer does provide options to export a ply either with mesh or without mesh (point cloud) right? I am actually trying to achieve the same mesh result from RealSense Viewer using pyrealsense. 

I did went through the RealSense SDK's **opencv_pointcloud_viewer.py** but what I need is actually the mesh, not the point cloud. 

I tried commenting out the spatial filter but it makes little to no difference to the FPS. I did see some issues #2157 regarding something similar but is it possible to create a pyrealsense2.frame class or changing the depth frame in the pyrealsense2.frame object? Not sure if only post process the depth frame and then align the processed depth frame and unprocessed color frame will provide a better result but currently I can't seem to find a way to do that. So, just wondering if this flow is possible?

closed 2022-12-05 by YanHernTan