realsenseai/librealsense #11142: Error when building librealsense from source with python3.11

chuyyy, 2022-11-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-131-generic                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.42.0                       | 

### Issue Description
I've run into an error when building librealsense from source with python3.11. The command used to build:

`cmake .. -DBUILD_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=true -DPYTHON_INCLUDE_DIRS=/usr/local/include/python3.11 -DPYTHON_LIBRARIES=/usr/local/lib/python3.11/site-packages

make install
`

Resulting error:

`[ 43%] Linking CXX shared library librealsense2.so
[ 43%] Built target realsense2
Scanning dependencies of target pybackend2
[ 43%] Building CXX object wrappers/python/CMakeFiles/pybackend2.dir/pybackend.cpp.o
In file included from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/attr.h:13,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/pybind11.h:45,
                 from /home/arb/py311-librealsense/wrappers/python/pybackend.cpp:4:
/home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h: In function ‘std::string pybind11::detail::error_string()’:
/home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h:446:36: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  446 |                 "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                    ^~
In file included from /usr/local/include/python3.11/Python.h:42,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/detail/common.h:122,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/pytypes.h:12,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h:13,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/attr.h:13,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/pybind11.h:45,
                 from /home/arb/py311-librealsense/wrappers/python/pybackend.cpp:4:
/usr/local/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
In file included from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/attr.h:13,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/pybind11.h:45,
                 from /home/arb/py311-librealsense/wrappers/python/pybackend.cpp:4:
/home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h:446:75: error: expected primary-expression before ‘>’ token
  446 |                 "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                                                           ^
/home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h:446:77: error: expected primary-expression before ‘)’ token
  446 |                 "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                                                             ^
/home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h:448:29: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  448 |                 handle(frame->f_code->co_name).cast<std::string>() + "\n";
      |                             ^~
In file included from /usr/local/include/python3.11/Python.h:42,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/detail/common.h:122,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/pytypes.h:12,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h:13,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/attr.h:13,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/pybind11.h:45,
                 from /home/arb/py311-librealsense/wrappers/python/pybackend.cpp:4:
/usr/local/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
In file included from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/attr.h:13,
                 from /home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/pybind11.h:45,
                 from /home/arb/py311-librealsense/wrappers/python/pybackend.cpp:4:
/home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h:448:64: error: expected primary-expression before ‘>’ token
  448 |                 handle(frame->f_code->co_name).cast<std::string>() + "\n";
      |                                                                ^
/home/arb/py311-librealsense/build/third-party/pybind11/include/pybind11/cast.h:448:66: error: expected primary-expression before ‘)’ token
  448 |                 handle(frame->f_code->co_name).cast<std::string>() + "\n";
[... 126 lines cut ...]

`[ 47%] Built target pybackend2
Scanning dependencies of target pyrealsense2
[ 47%] Building CXX object wrappers/python/CMakeFiles/pyrealsense2.dir/python.cpp.o
[ 47%] Building CXX object wrappers/python/CMakeFiles/pyrealsense2.dir/c_files.cpp.o
[ 48%] Building CXX object wrappers/python/CMakeFiles/pyrealsense2.dir/pyrs_advanced_mode.cpp.o
[ 48%] Building CXX object wrappers/python/CMakeFiles/pyrealsense2.dir/pyrs_context.cpp.o
[ 48%] Building CXX object wrappers/python/CMakeFiles/pyrealsense2.dir/pyrs_device.cpp.o
/home/arb/py311-librealsense/wrappers/python/pyrs_device.cpp: In lambda function:
/home/arb/py311-librealsense/wrappers/python/pyrs_device.cpp:123:34: error: ‘cerr’ is not a member of ‘std’
  123 |                             std::cerr << "?!?!?!!? exception in python register_calibration_change_callback ?!?!?!?!?" << std::endl;
      |                                  ^~~~
/home/arb/py311-librealsense/wrappers/python/pyrs_device.cpp:8:1: note: ‘std::cerr’ is defined in header ‘<iostream>’; did you forget to ‘#include <iostream>’?
    7 | #include "../include/librealsense2/hpp/rs_record_playback.hpp" // for downcasts
  +++ |+#include <iostream>
    8 | //#include <iostream>
make[2]: *** [wrappers/python/CMakeFiles/pyrealsense2.dir/build.make:115: wrappers/python/CMakeFiles/pyrealsense2.dir/pyrs_device.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1120: wrappers/python/CMakeFiles/pyrealsense2.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`

I tried the suggested fix (adding #include <iostream> in librealsense/wrappers/python/pyrs_device.cpp) and the build was successful. I don't know if this fix masked some other unknown issue. Your input would be greatly appreciated.

Other similar issues:
- seems like [some changes](https://github.com/sirfz/tesserocr/issues/298#issuecomment-1100035852) were made to the PyFrameObject.

MartyG-RealSense, 2022-11-25:
Thanks so much @chuyyy for your report of the solution that worked for you!

@maloel  Would you be able to assist @chuyyy with their questions at the bottom of this issue, please?  Thanks!

MartyG-RealSense, 2022-12-04:
@maloel Would you be able to assist @chuyyy with their questions in this issue please? Thanks!

maloel, 2022-12-04:
Hi @chuyyy 
Thank you for the detailed report and fix.
Your instincts are good, and the changes you made seem fine, though we cannot make those changes ourselves yet:
* Pybind 2.10.1 is the first version to fully support embedding the newly released Python 3.11
* I have recently upgraded our internal version to 2.9.0 (2.9.2 fails)
* Version 2.10 is no longer compatible with MSVC 2015, which is currently what we use in our CI (for various reasons), therefore we cannot make this change yet, and won't support the new Python until we do
* I do have it on my list of things to change, and you should see many such changes coming in 2023

chuyyy, 2022-12-09:
Thanks for the update @maloel. Would you be able to provide a more specific timeline for when this will be ready or is there a ticket that I can track?

MAVProxyUser, 2022-12-26:
This librealsense python3 / pybind11 issue is for the birds! Really annoying.

maloel, 2022-12-27:
Our next release is scheduled for some time towards the end of the first quarter, but I have no idea if support for 3.11 will make it in there.
I'll bring your feedback to our priority review for this release :)
Cheers and happy New Year!

MAVProxyUser, 2022-12-27:
Cheers, and thank you for your diligence. 

For anyone else following along, this worked for me: 
```
diff --git a/third-party/pybind11/CMakeLists.txt b/third-party/pybind11/CMakeLists.txt
index eae671d8c..ed0cc31d3 100644
--- a/third-party/pybind11/CMakeLists.txt
+++ b/third-party/pybind11/CMakeLists.txt
@@ -8,7 +8,7 @@ ExternalProject_Add(
         pybind11
         PREFIX .
         GIT_REPOSITORY "https://github.com/pybind/pybind11.git"
-        GIT_TAG "8de7772cc72daca8e947b79b83fea46214931604" # v2.6.2
+        GIT_TAG "80dc998efced8ceb2be59756668a7e90e8bef917" # v2.10.1
         SOURCE_DIR "${CMAKE_BINARY_DIR}/third-party/pybind11"
         # Override default steps with no action, we just want the clone step.
         CONFIGURE_COMMAND ""
diff --git a/wrappers/python/pyrs_device.cpp b/wrappers/python/pyrs_device.cpp
index 711678171..79e4a022b 100644
--- a/wrappers/python/pyrs_device.cpp
+++ b/wrappers/python/pyrs_device.cpp
@@ -6,6 +6,7 @@ Copyright(c) 2017 Intel Corporation. All Rights Reserved. */
 #include <librealsense2/hpp/rs_device.hpp>
 #include <librealsense2/hpp/rs_record_playback.hpp> // for downcasts
 #include <../common/metadata-helper.h>
+#include <iostream>
 
 void init_device(py::module &m) {
     /** rs_device.hpp **/
```

This is the compile command that worked
```
+cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_WITH_CUDA=true -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.8 -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_SHARED_LIBS=false
```

I also forcibly removed python2 
```
sudo apt-get remove python2.7 python2
```

MartyG-RealSense, 2022-12-31:
Thanks very much @chuyyy and @MAVProxyUser for sharing your solutions with the RealSense community!  As your problems have been resolved, I will close this case.  Thanks again!

closed 2022-12-31 by MartyG-RealSense