realsenseai/librealsense #11143: The source directory "/home/dep" does not appear to contain CMakeLists.txt. please quote a solution

Sashraa, 2022-11-24:
The standard `cmake` technique is to add the location of `librealsense` to your `CMAKE_PREFIX_PATH` with something like this (run in your build directory):
```sh
cmake -DCMAKE_PREFIX_PATH="C:\Program Files (x86)\Intel RealSense SDK 2.0" .
```
and then to ask `cmake` to find `librealsense` using that path in your `CMakeLists.txt` like so
```cmake
find_package(realsense2)

target_link_libraries(${PROJECT_NAME} PRIVATE realsense2::realsense2)
```
this should deal with the include paths, finding the DLL, etc. automatically.

_Originally posted by @radfordi in https://github.com/IntelRealSense/librealsense/issues/4277#issuecomment-505242118_

Sashraa, 2022-11-24:
E: Unable to locate package librealsense2-dkms
 
Please suggest solutions for it

MartyG-RealSense, 2022-11-24:
Hi @Sashraa  As you are quoting https://github.com/IntelRealSense/librealsense/issues/4277#issuecomment-505242118 it sounds as though you are attempting to compile a project that you have created yourself.  Is that correct, please?

If it is then the **Getting Started** project at the link below may be a helpful reference.  

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

It is the **rs-hello-realsense** RealSense SDK example program but with a more complex CMakeLists.txt file that has the instructions mentioned in https://github.com/IntelRealSense/librealsense/issues/4277#issuecomment-505242118 that enable the project to find the RealSense libraries.

MartyG-RealSense, 2022-11-24:
In regard to the error **E: Unable to locate package librealsense2-dkms** - were you attempting to install the librealsense SDK when this error occurred?  If you were then what installation instructions were you following, please?

Sashraa, 2022-11-24:
> Hi @Sashraa As you are quoting [#4277 (comment)](https://github.com/IntelRealSense/librealsense/issues/4277#issuecomment-505242118) it sounds as though you are attempting to compile a project that you have created yourself. Is that correct, please?
> 
> If it is then the **Getting Started** project at the link below may be a helpful reference.
> 
> https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started
> 
> It is the **rs-hello-realsense** RealSense SDK example program but with a more complex CMakeLists.txt file that has the instructions that enable the project to find the RealSense libraries and build the project as a standalone executable.
> 
> Can you confirm please whether you are using Windows or Linux? The quoted prefix path `cmake -DCMAKE_PREFIX_PATH="C:\Program Files (x86)\Intel RealSense SDK 2.0"` is for a folder location on a Windows computer. The title of the issue about /home/dep sounds as though you are using a Linux computer.

I have solved this one referring to your previous solutions
Thank you.

Sashraa, 2022-11-24:
I am not able to access opencv I installed in the home directory.. Kindly suggest links for installing opencv in the correct manner.

MartyG-RealSense, 2022-12-02:
Hi @Sashraa  Do you require further assistance with this case, please?  Thanks!

Sashraa, 2022-12-08:
hello,
 rs2::pipeline _pipe;
rs2::config _cfg;
       _cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
        _pipe.start(_cfg);

         rs2::frameset cframes=_pipe.wait_for_frames();
there is an error in the execution at this particular line: rs2::frameset cframes=_pipe.wait_for_frames(); 
wait_for_frames() seems to not work...kindly assist.

Sashraa, 2022-12-08:
I have retrieved IR and depth frames in .png but I am not able to retrieve the color image due to the above issue.. kindly assist.

MartyG-RealSense, 2022-12-08:
What is the error message that you receive, please?

Sashraa, 2022-12-09:
[image]

Sashraa, 2022-12-09:
it's simply an error...there are frames captured

Sashraa, 2022-12-09:
if there are any methods to retrieve the color image ..kindly share

MartyG-RealSense, 2022-12-09:
Does the script work if you do it like this:

 ```
 rs2::frameset cframes;
    for(int i = 0; i < 30; i++)
    {
  
        cframes = _pipe.wait_for_frames();
    }

    rs2::frame color_frame = cframes.get_color_frame();
```

Sashraa, 2022-12-09:
No, it doesn't work...I am using D410 sensor.
Can color image be retrieved in png format?

Sashraa, 2022-12-09:
color image for gender classification

MartyG-RealSense, 2022-12-09:
If you are using a D410 Depth Module Kit then it will not be equipped with an RGB sensor and so will not support the color stream format.

You could try obtaining RGB from the **left infrared** sensor though.

_cfg.enable_stream(RS2_STREAM_INFRARED, 640, 480, RS2_FORMAT_BGR8, 30);

Sashraa, 2022-12-10:
Thanks a lot for your guidance...

MartyG-RealSense, 2022-12-12:
The RealSense **OpenVINO Toolkit** wrapper for C++ supports object detection and gender / age detection in its **Face** and **DNN** examples.  However, both those examples use the color stream.  And because they use depth to color alignment, it does not seem to be practical to adapt them to use RGB from the infrared stream.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino

Sashraa, 2022-12-13:
Thank you.

Imshow doesn't work.. what do I do?

MartyG-RealSense, 2022-12-13:
Are you using an OpenVINO example program that uses OpenCV, please (imshow is an OpenCV instruction).

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino#getting-started

Sashraa, 2022-12-13:
no I am using opencv example only...

MartyG-RealSense, 2022-12-13:
Thank you.  Which example are you using?

Sashraa, 2022-12-13:
Th rs-align example

MartyG-RealSense, 2022-12-13:
When you said "imshow doesn't work", do you mean the OpenCV **imshow** example.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/imshow

If you are using that example then it does not make use of color or infrared, only depth.  It **color shades** the depth coordinates and does not use any other type of stream.

But if you are trying to align depth to infrared then that will not work with these OpenCV examples.  This is because you can only use an alignment instruction to align the depth stream to the color stream, or the color stream to the depth stream.

Sashraa, 2022-12-13:
okay..got it... Thank you soo much.

Sashraa, 2022-12-14:
hello, I've installed the opencv v.4.6... what is the content I must add in the pkgconfig opencv.pc file?

Sashraa, 2022-12-15:
Unable to init server: Could not connect: Connection refused
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.6.0) /home/me/opencv/modules/highgui/src/window_gtk.cpp:635: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'

This is the error I get when I use imshow.

this is the code I used.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/imshow/rs-imshow.cpp

MartyG-RealSense, 2022-12-15:
The answer at the link below seems to fit your situation with imshow.

https://www.reddit.com/r/opencv/comments/rb8z0q/bug_problems_with_opencv_and_raspberrypi_imshow/

Sashraa, 2022-12-16:
thank you....
I've got an issue with the realsense viewer... This is the op I get...
[image]
It's not streaming... will there be streaming if I connect it to a monitor...? Is there an option to enable RGB in D410?

MartyG-RealSense, 2022-12-16:
As you mention about whether it would work if you attached a monitor, this suggests to me that you are using a 'headless' computer setup that does not have a display.  Is that correct, please?

If you are using graphical examples and tools such as realsense-viewer then a display should be attached, otherwise there may be problems if your librealsense source-code build includes support for graphics (which it does by default) but a display is not attached.  In Intel's build configuration guidance (at the link below) it is recommended to include the build flag **-DBUILD_GRAPHICAL_EXAMPLES=false** in your CMake build instruction if your system is headless, though graphical tools such as realsense-viewer will not be installed as a consequence of doing so.

https://dev.intelrealsense.com/docs/build-configuration

[image]

If DBUILD_GRAPHICAL_EXAMPLES is false then graphicless text console tools and examples should still be built because the flag **DBUILD_EXAMPLES** is true by default.

Sashraa, 2022-12-16:
yeah I am using headless...is this the reason for imshow not working?

MartyG-RealSense, 2022-12-16:
Yes, I believe that it is expecting a display to be attached for a graphical function like imshow.

Sashraa, 2022-12-22:
Nope! Thank you soo much for your guidance.

closed 2022-12-22 by Sashraa

MartyG-RealSense, 2022-12-22:
You are very welcome, @Sashraa - thanks very much for the update!

Sashraa, 2023-01-06:
Hello I am trying to create a Http server client communication...
while compiling my code.. I keep getting the below errors:
undefined references to `gstProxySettings

MartyG-RealSense, 2023-01-06:
I cannot find any references to the term **gstProxySettings**.  What method are you using to set up your http connection please?

Sashraa, 2023-03-27:
Hello,
I have been trying to install Librealsense through all the methods.. but it fails in the make.. I have tried the debian, RSUSB methods and the Libusb backend version in Banana Pi. 
Kernel Version: 4.9.241-BPI-M5
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

Is there support for Librealsense in BananaPi? or please suggest any other methods ..
Kindly support..
Thank You.

MartyG-RealSense, 2023-03-27:
There are known problems with installing librealsense on Raspian Bullseye.  The previous version, Raspbian **Buster**, usually works better.

 If you would prefer to use Bullseye, at https://github.com/IntelRealSense/librealsense/issues/9943#issuecomment-966048785 a RealSense user describes their method for adjusting Bullseye for librealsense compatibility by copying two binaries from Buster to Bullseye.

Sashraa, 2023-03-27:
Thank you soo much...It is of a great help