realsenseai/librealsense #11148: Slow performance in Windows 11

LostCanadian, 2022-11-24:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.13.00.50 | 
| Operating System & Version | Win 11 | 
| Platform                                 | PC |
| SDK Version                            | 2.51.1 | 
| Language                            | C++ | 
| Segment			| 3D Scanning | 

### Issue Description
We have been encountering major performance issues with RealSense in Windows 11. We first noticed it in our 3D scanning software product which was running at ~12 Hz instead of the usual 30 Hz, but it's also apparent in the RealSense Viewer, with frame rates varying wildly. While it's not 100% consistent, in general it appears to happen more often with the latest 22H2 build of Windows 11.

We typically set up our customers with a Surface Pro, but the Surface Pro 9 with Windows 11 is almost unusable. We experimented with installing Windows 10 on the Surface Pro 9, and performance is fast, so it seems like some sort of compatibility issue with Win 11. We are also experiencing issues on Win 11 desktop PCs, although some PCs seem to be OK. It's pretty frustrating, since we are getting inundated with customer reports of slow performance.

My initial thoughts are that this may be a Microsoft issue, not an Intel one, but I wanted to post it here to see if anyone else has encountered this and perhaps has some ideas on fixes or workarounds.

MartyG-RealSense, 2022-11-25:
Thanks very much @LostCanadian for the report!  Other RealSense users of Windows 11 are welcomed to contribute details of any slow-down that they are experiencing on their own Windows 11 computers,

MartyG-RealSense, 2022-12-08:
I am not aware of a RealSense team member with a Windows 11 system containng that update.  If you could supply test feedback then it would be welcome.  Thanks!

LostCanadian, 2022-12-20:
No, I was hoping to test out the aforementioned fix, but I can't seem to get it to show up in Windows Update, even under optional updates. It's tough to tell for sure, but it sounds like those fixes will be on hold from the main update stream until sometime in January.

Until we can get the update from Microsoft, we're a bit stuck. For now, we are actually recommending that customers using a Surface Pro 9 should special order them with Windows 10 installed due to the performance issues with RealSense and Windows 11. That said, given that we are operating under the impression that this is a Microsoft issue, not an Intel one, I can close this ticket for now.

If it turns out that we still have performance issues later in 2023, I will connect with you guys again.

closed 2022-12-20 by LostCanadian

MartyG-RealSense, 2022-12-20:
Thanks very much @LostCanadian for the update!

LostCanadian, 2023-02-13:
Just to close the loop on this in case anyone else runs into this - we eventually narrowed it down to an issue with the way Windows 11 deprioritizes background processes, notably on CPUs which have "efficiency cores". Our GUI application runs in a separate process from our RealSense capture and 3D rendering code; our theory is that Windows 11 is (unnecessarily) boosting priority of the (foreground) GUI process and deprioritizing our "background" process which actually does most of the heavy lifting for our app. Setting the priority for the process to "Above Normal" resolves all of our performance issues. In code, that amounts to a "SetPriorityClass()" function call when the process starts up.

Typically we try to avoid changing the process priority, since it generally seems to be bad etiquette. In this case however, it appears to be appropriate/needed due to the poor handling of processes in Windows 11 with the newer Intel CPUs.

MartyG-RealSense, 2023-02-14:
Thanks so much @LostCanadian for sharing your excellently detailed Windows 11 experience!