realsenseai/librealsense #11151: Setting up additional D435i inside LXC container on Jetson (Ubuntu 18.04/22.04)

martinrebane, 2022-11-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 18.04 (host) Ubuntu 22.04 (lxd)
| Kernel Version (Linux Only)    |  4.9.140+ | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            |  2.41 (host) 2.51.1 (lxd) | 
| Language                            |  ROS/python                       | 
| Segment			|  Robot                         | 

### Issue Description

I would like to ask guidance on some issues setting up additional RealSense camera using lxd container. I am running Jetson TX2 which has Ubuntu 18.04 installed and several RealSense modules. Now I would like to add additional RealSense D435i camera and run it using ROS2 in a lxc container using Ubuntu 22.04 (so I can use newer software, I am not able to upgrade the main system, so lxc is the only option I can think about).

My questions.

1) Will it be possible at all? Will librealsense on the host capture and interfere with the new camera (it also handles other realsense cameras which should keep working) even when I am not programming it in any way to handle the new camera? The host has librealsense udev rules applied.

2) librealsense on host machine is compiled with `RSUSB`. Should I compile librealsense in the container also with `-DFORCE_RSUSB_BACKEND` or is it possible to use native kernel drivers? As lxc container uses the kernel of the host, lxc should also use RSUSB?

3) Which devices should be mapped to to the container? Only the USB camera or also /dev/videoX devices?

4) Is it OK to use newer librealsense (2.51.1) inside lxc? The host has 2.41.

5) Any other suggestions I should consider?

MartyG-RealSense, 2022-11-26:
Hi @martinrebane  There is no information available unfortunately regarding using librealsense or the RealSense ROS wrapper with LXC / LXD.  A similar alternative approach that is better documented would be **Docker**.  

With Docker it is possible to use different librealsense SDK versions on the host and the Docker container.

MartyG-RealSense, 2022-12-04:
Hi @martinrebane  Do you require further assistance with this case, please?  Thanks!

martinrebane, 2022-12-04:
Thanks for your input! I managed to set up RealSense successfully using LXC. It is very easy once you figure out how to do it and I'd actually encourage anyone to use LXC in robotics instead of docker. Escpecially as it can be stateful (unlike docker which is foremost meant for stateless applications, e.g., for web services). And LXC uses host kernel, which is another advantage when dealing with hardware.

So, here we go.

1. Set up LXC and launch a suitable container. You need to install  a) librealsense b) ros and c) realsense-ros wrapper in the container (of course, use ROS only if you need ROS). As of Dec 2022 max Ubuntu they all officially support is still 20.04 (use ROS2 Galactic). For easiest setup go for it, use apt-get and follow official tutorials. However, if you are ready to compile your own software, I successfully managed to set up and compile  ROS2 Humble too on one of the computers which also worked well with D435 camera.
2. (Optional step). In case you used native backend for librealsense (this is the default) you are able to check (both on the host and from the container) whether the camera is found:
```
sudo apt install v4l-utils
v4l2-ctl --all
```
It should list your Realsense there. Do not continue if your host does not display your camera, fix it first. NB! It will not display it for RSUSB backend, this is OK!
3. I guess most people make it this far themselves. So here comes the important stuff. Now map your camera to your container (run this on host):
```
lxc config device add <container_name> <device_name_for_realsense> usb vendorid=8086 productid=0b07 gid=1000
```
`<container_name>` is of course the name of your lxc container, `<device_name_for_realsense>` a name you assign for your device, just make it up. `usb vendor` value is `8086` (intel) so no need to change this. You might need to change `productid`. For my D435 it is `0b07`, but other models (including D435i) have different ID. You can see yours with `lsusb` command. The most important part here is `gid` (your group id which you can see with `id` command on the host). It gives the container permissions to use the device. Otherwise you will just map it but cannot use it as you have no permissions. You can read more about permissions [here](https://ubuntu.com/blog/custom-user-mappings-in-lxd-containers). However, the tutorial is old, so they do not show `gid=` option there, but the logic is still the same.

OK. If you used RSUSB backend then you should be done, everything should be working after you restart the container. Launch your realsense-ros wrappers's node to check that it works. NB! With RSUSB I got several strange warnings which did not seem to affect the node and everything I needed was working, camera topics were all published. The warning is this:
```
[realsense2_camera_node-1]  04/12 12:00:18,226 WARNING [547373427104] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```
@MartyG-RealSense Perhaps you can comment what might be causing it?

4. If you are using native USB backend (the default option) then you will also need to map `/dev/videoX` devices the same way. For each device (in my case video0 ... video5, but check yours with `ls -la /dev/video*` on host) do (replace X in both places):
```
lxc config device add <container_name> videoX unix-char path=/dev/videoX gid=1000
```
5. Restart your container with `lxc restart <container_name>` and you should be good to go. Test it out:
```
cd ros2_ws/
source install/setup.bash
ros2 launch realsense2_camera rs_launch.py
```
Have fun! :)

MartyG-RealSense, 2022-12-04:
Thanks so much @martinrebane for your invaluable highly detailed LXC setup guide!

The control_transfer returned warning type **Resource temporarily unavailable** typically indicates a serious communication problem with the device if it generates continuously in the log, but can be safely ignored if it only occurs a small number of times during launch and then stops, or occurs a couple of times per minute after launch.

Even if a control_transfer returned warning generates continuously in the log after launch, it can be ignored if the topics are publishing normally in RViz.

martinrebane, 2022-12-04:
Thanks @MartyG-RealSense for a quick response! This warning immediately appears a few times after launching the node and then approximately 10 times per hour. So I guess this can be ignored :)

closed 2022-12-04 by martinrebane