realsenseai/librealsense #11153: The Custom Intrinsic written by using CustomRW is changed arbitrarily!

demul, 2022-11-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | CustomRW for Intel RealSense D400, Version: 2.12.0.0  | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  5.4.0-1060-raspi | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  CustomRW for Intel RealSense D400, Version: 2.12.0.0 }  | 
| Segment			|  {Robot}  | 

### Issue Description
When I try to write my own intrinsic by 
`Intel.Realsense.CustomRW -w -f my_intrinsic.xml`
then
`Intel.Realsense.CustomRW -r`
I can see below
`CustomRW for Intel RealSense D400, Version: 2.12.0.0

  Device PID: 0B5C
  Device name: Intel RealSense D455
  Serial number: 215122255233
  Firmware version: 05.13.00.50
  USB Type: 3.2

reading calibration parameters from input file ...
Calbration data updated successfully.
Calibration parameters from the device:
  resolutionLeftRight: 1280 800

  FocalLengthLeft: 1000.000000 1000.000000
  PrincipalPointLeft: 637.592957 422.867493
  DistortionLeft: -0.054713 0.078974 0.000598 0.000707 -0.026655

  FocalLengthRight: 1000.000000 1000.000000
  PrincipalPointRight: 643.581787 392.125000
  DistortionRight: -0.051063 0.062103 0.000188 0.000143 -0.019482

  RotationLeftRight: 0.999919 0.001671 -0.012587
                     -0.001579 0.999972 0.007337
                     0.012599 -0.007316 0.999894
  TranslationLeftRight: -95.065918 -0.239313 -0.584865

  HasRGB: 1

  resolutionRGB: 1280 800

  FocalLengthColor: 1000.000000 1000.000000
  PrincipalPointColor: 640.000000 400.000000
  DistortionColor: 0.000000 0.000000 0.000000 0.000000 0.000000
  RotationLeftColor: 1.000000 0.000000 0.000000
                     0.000000 1.000000 0.000000
                     0.000000 0.000000 1.000000
  TranslationLeftColor: 0.000000 0.000000 0.000000`

and when I turn on ros librealsense wrapper and turn off it and run below
`Intel.Realsense.CustomRW -r`
The focal length is changed!! (Principal point and Distortion and Transform is not changed. Just focal length is changed arbitrarily)
`CustomRW for Intel RealSense D400, Version: 2.12.0.0

  Device PID: 0B5C
  Device name: Intel RealSense D455
  Serial number: 215122255233
  Firmware version: 05.13.00.50
  USB Type: 3.2

Calibration parameters from the device:
  resolutionLeftRight: 1280 800

  FocalLengthLeft: 1008.444702 1008.444702
  PrincipalPointLeft: 637.592957 422.867493
  DistortionLeft: -0.054713 0.078974 0.000598 0.000707 -0.026655

  FocalLengthRight: 1008.444702 1008.444702
  PrincipalPointRight: 643.581787 392.125000
  DistortionRight: -0.051063 0.062103 0.000188 0.000143 -0.019482

  RotationLeftRight: 0.999919 0.001671 -0.012587
                     -0.001579 0.999972 0.007337
                     0.012599 -0.007316 0.999894
  TranslationLeftRight: -95.065918 -0.239313 -0.584865

  HasRGB: 1

  resolutionRGB: 1280 800

  FocalLengthColor: 1008.444702 1008.444702
  PrincipalPointColor: 640.000000 400.000000
  DistortionColor: 0.000000 0.000000 0.000000 0.000000 0.000000
  RotationLeftColor: 1.000000 0.000000 0.000000
                     0.000000 1.000000 0.000000
                     0.000000 0.000000 1.000000
  TranslationLeftColor: 0.000000 0.000000 0.000000`

Why the focal length are changed?

and when I repeatedly turn on and off ros librealsense(without writing), the focal length is changed about 3~4 times.
But after 3~4 times of turning on/off, the focal length is fixed and is not changed anymore.
What is the problem?! I cannot understand this phenomenon. It's crazy!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
plz help me!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

ex : The focal length value(1000.0) is jsut dummy to show this phenomenon clearly.

MartyG-RealSense, 2022-11-28:
Hi @demul  This does not sound like something to be concerned about.  Changing the focal length values in the calibration table will not affect the focus of the camera, as the focus of the camera's lenses is fixed and cannot be changed with software, as confirmed by RealSense team members at https://github.com/IntelRealSense/librealsense/issues/7814#issuecomment-732772348

demul, 2022-11-29:
@MartyG-RealSense 
Thanks to u matry! I'm relieved.
but I'm just curious why calibration table is changed.
Can u explain???

MartyG-RealSense, 2022-11-29:
I have not seen a change in focal length in ROS being previously reported by RealSense users on this support forum.  I did find such a report on another website though in which a RealSense user reported that it was caused by applying a Decimation post-processing filter.  Does your own ROS launch use the Decimation filter, please?

[image]

demul, 2022-11-30:
@MartyG-RealSense 
Thanks to you marty!
It seem that the decimation filter is applied on my launcher.

So, when the ROS launcher apply decimation filter, the intrinsic on chip is automatically changed(using SDK API) along with the resolution change caused by decimation. is it correct?

MartyG-RealSense, 2022-11-30:
A resolution change will alter fx and fy.  And focal length is a multiple of fx and fy.  I am not completely certain if the decimation filter is the cause though.  The best way to test this would be to remove the decimation filter from your launch and see if the focal length stops changing size when ROS is launched several times.

demul, 2022-11-30:
after Thinking about it carefully, but I think your explaination makes no sense at all.
The change in resolution **scale focal length** and **move principal point**.
but as You can see, The principal point is not changed at all after ROS lanched. Only focal length is changed.
and even The focal length is increased! while the decimation downscale image resolution.

and The focal length from `Intel.Realsense.CustomRW -r` is the focal length about the fixed resolution Y16 stream.
If the focus of the camera's lenses is fixed and cannot be changed with software as you told, There must be not any changes on the focal lenght from `Intel.Realsense.CustomRW -r`.

So your explaination which applying a decimation post-processing filter can cause focal length change of the intrinsic on chip is incorrect.
so I think more attention is needed for this case. It's completely incomprehensible!!!!!!!!!!!!!

MartyG-RealSense, 2022-11-30:
I did say that I was not certain that the decimation filter was the cause and offered the alternative possibility that the camera's USB connection might be being mis-detected as USB 2.1 instead of USB 3.  You can check whether this is happening by looking at the **Device USB type** line on each launch's log and seeing if it has the value '2.1' or '3.2' beside it.

demul, 2022-11-30:
Thank you marty!

We use USB3.0. But the launcher mis-detect it as USB3.2.
and even `Intel.Realsense.CustomRW -r` mis-detect it as USB3.2.
Can it be the cause of the problem??

And, by the way, i figured out "Turning on depth stream is the condition of that focal length change".
Turning on only infra camera do not change that focal length.

MartyG-RealSense, 2022-11-30:
USB 3.0 and USB 3.2 are both USB 3, so it should not result in a change in the resolutions supported by the camera.  So the focal length problem does not seem to be due to a mis-detection of the USB connection type.

If the focal length change does not have a noticable effect on the images that are produced by the ROS topics and they perform well in RViz then it is likely to be safe to ignore it even if you do not find out for certain what the cause was.

MartyG-RealSense, 2022-12-06:
Hi @demul  Do you require further assistance with this case, please?  Thanks!

demul, 2022-12-07:
@MartyG-RealSense 
I'm sorry for the late response.
Frankly, I found out this problem during investigating another problem about realsense.
I'm going to make a new issue of the problem originally i want to solve.
Maybe, This problem (arbitrary changing of custom intrinsic written by CustomRW) just looks fine.
I'll close this issue and keep going on new issue!
Thanks!

closed 2022-12-07 by demul

MartyG-RealSense, 2022-12-07:
No need for apologies.  :)  Thanks very much @demul for the update!