realsenseai/librealsense #11154: Convert Rosbag file into pointcloud file

TheNemo05, 2022-11-28:
I ve recorded the video sequence using realsense SDK and it is stored in the .bag format. The RS convert tool didnt help as after conversion the file location is unknown. Can someone get me a valid solution.

MartyG-RealSense, 2022-11-28:
Hi @TheNemo05  Any script originally designed for use with a live camera can use a bag file as its data source instead by inserting an **enable_device_from_file** instruction before the pipeline start line.   For example, to use a bag file called 'test.bag' as the data source using C++ code:

```
rs2::config cfg;
cfg.enable_device_from_file("test.bag");
pipeline.start(cfg);
```

An example of a Python equivalent would be:

```
cfg = rs.config()
rs.cfg.enable_device_from_file(cfg,'test.bag')
pipeline.start(cfg)
```

**********

Once you know how to adapt a live-camera script to a bag-reading script using the above method then you can easily adapt existing pointcloud generation scripts.

If you wish to obtain a **.ply** pointcloud data file then adapting a ply export script will likely be the best course of action.

The RealSense SDK has a ply export example program called **export_ply_example**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

A C++ example of a ply export script is at https://github.com/IntelRealSense/librealsense/issues/6677#issuecomment-1101712660

TheNemo05, 2022-11-29:
hello how do we use the export_ply_example with realsense SDK, So I could receive exported file in .ply format.

MartyG-RealSense, 2022-11-29:
To use the SDK's Python example scripts it is necessary to install the SDK's **pyrealsense2** wrapper.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

If you have a desktop or laptop PC computer then the easiest method for installing the wrapper is with pip packages using the instruction **pip install pyrealsense2** though this method cannot be used with computers with an Arm architecture such as Raspberry Pi and Nvidia Jetson.

The other way to install the wrapper is to build it from source code with CMake.  You can either build it from source after the librealsense SDK has already been installed, or build librealsense and the wrapper at the same time.