realsenseai/librealsense #11156: Usbcore kernel error

Allius27, 2022-11-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.07.150  | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.4.0-81-generic  | 
| Platform                                 | NUC10i5FNK  |
| SDK Version                            |  2.50  | 
| Language                            |  c++  | 
| Segment			|  others  | 

Hi there. 
We changed sdk from version 2.39 to 2.50 on ubuntu16

And now we had extremely strange situation.
Our video is looping, after some experiments we decided, that the reason might be the camera or kernels in our system.

dmesg logs on system startup
 
<details>
  <summary>Case 1</summary>

```
[   11.147814] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[   11.149761] usbcore: registered new interface driver btusb
[   11.149864] uvcvideo: Unable to create debugfs 2-2 directory.
[   11.149927] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input3
[   11.149969] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[   11.150725] uvcvideo: Unable to create debugfs 2-2 directory.
[   11.150826] usbcore: registered new interface driver uvcvideo
[   11.150829] USB Video Class driver (1.1.1)
[   11.177103] ------------[ cut here ]------------
[   11.177111] WARNING: CPU: 5 PID: 633 at /build/linux-cs3yMe/linux-4.4.0/sound/hda/hdac_i915.c:191 snd_hdac_i915_register_notifier+0x2d/0x40 [snd_hda_core]()
[   11.177112] Modules linked in: snd_hda_codec_hdmi(+) kvm snd_hda_intel irqbypass snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_timer snd soundcore btusb btrtl uvcvideo(OE) btbcm videobuf2_vmalloc btintel videobuf2_memops videobuf2_v4l2 bluetooth videobuf2_core v4l2_common shpchp videodev(OE) media ite_cir rc_core tpm_crb(+) acpi_als kfifo_buf acpi_pad mac_hid industrialio ip6t_REJECT nf_reject_ipv6 xt_hl nf_conntrack_ipv6 nf_defrag_ipv6 ip6t_rt ipt_REJECT nf_reject_ipv4 xt_limit xt_tcpudp nf_conntrack_ipv4 nf_defrag_ipv4 xt_addrtype xt_conntrack ip6table_filter ip6_tables ib_iser nf_conntrack_netbios_ns nf_conntrack_broadcast rdma_cm iw_cm nf_nat_ftp nf_nat ib_cm ib_sa nf_conntrack_ftp ib_mad nf_conntrack ib_core iptable_filter ib_addr ip_tables iscsi_tcp x_tables libiscsi_tcp libiscsi scsi_transport_iscsi
[   11.177139]  autofs4 btrfs raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear uas usb_storage crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul glue_helper ablk_helper e1000e(OE) cryptd ptp sdhci_pci pps_core sdhci nvme ahci libahci wmi i2c_hid hid video fjes
[   11.177158] CPU: 5 PID: 633 Comm: modprobe Tainted: G           OE   4.4.0-81-generic #104-Ubuntu
[   11.177160] Hardware name: Intel(R) Client Systems NUC10i5FNK/NUC10i5FNB, BIOS FNCML357.0052.2021.0409.1144 04/09/2021
[   11.177161]  0000000000000286 00000000b652085d ffff88017af03a60 ffffffff813f9683
[   11.177163]  0000000000000000 ffffffffc0632d70 ffff88017af03a98 ffffffff81081322
[   11.177165]  ffff880177e96800 0000000000000008 0000000000000007 0000000000000009
[   11.177166] Call Trace:
[   11.177171]  [<ffffffff813f9683>] dump_stack+0x63/0x90
[   11.177173]  [<ffffffff81081322>] warn_slowpath_common+0x82/0xc0
[   11.177175]  [<ffffffff8108146a>] warn_slowpath_null+0x1a/0x20
[   11.177179]  [<ffffffffc063082d>] snd_hdac_i915_register_notifier+0x2d/0x40 [snd_hda_core]
[   11.177183]  [<ffffffffc04e209d>] patch_generic_hdmi+0x5bd/0x6b0 [snd_hda_codec_hdmi]
[   11.177187]  [<ffffffff8157c241>] ? __regmap_init+0xa91/0xac0
[   11.177194]  [<ffffffffc064845b>] hda_codec_driver_probe+0x6b/0x100 [snd_hda_codec]
[   11.177197]  [<ffffffff81562ee2>] driver_probe_device+0x222/0x4a0
[   11.177198]  [<ffffffff815631e4>] __driver_attach+0x84/0x90
[   11.177200]  [<ffffffff81563160>] ? driver_probe_device+0x4a0/0x4a0
[   11.177202]  [<ffffffff81560b0c>] bus_for_each_dev+0x6c/0xc0
[   11.177204]  [<ffffffff8156269e>] driver_attach+0x1e/0x20
[   11.177206]  [<ffffffff815621db>] bus_add_driver+0x1eb/0x280
[   11.177207]  [<ffffffffc0442000>] ? 0xffffffffc0442000
[   11.177209]  [<ffffffff81563af0>] driver_register+0x60/0xe0
[   11.177213]  [<ffffffffc06480da>] __hda_codec_driver_register+0x5a/0x60 [snd_hda_codec]
[   11.177216]  [<ffffffffc044201e>] hdmi_driver_init+0x1e/0x1000 [snd_hda_codec_hdmi]
[   11.177218]  [<ffffffff81002123>] do_one_initcall+0xb3/0x200
[   11.177221]  [<ffffffff811d1511>] ? __vunmap+0x91/0xe0
[   11.177223]  [<ffffffff811ed723>] ? kmem_cache_alloc_trace+0x183/0x1f0
[   11.177225]  [<ffffffff811ee51a>] ? kfree+0x13a/0x150
[   11.177228]  [<ffffffff8118dd23>] do_init_module+0x5f/0x1cf
[   11.177230]  [<ffffffff8110aaaf>] load_module+0x166f/0x1c10
[   11.177232]  [<ffffffff81107050>] ? __symbol_put+0x60/0x60
[   11.177234]  [<ffffffff812153b0>] ? kernel_read+0x50/0x80
[   11.177236]  [<ffffffff8110b294>] SYSC_finit_module+0xb4/0xe0
[   11.177238]  [<ffffffff8110b2de>] SyS_finit_module+0xe/0x10
[   11.177240]  [<ffffffff81840a72>] entry_SYSCALL_64_fastpath+0x16/0x71
[   11.177242] ---[ end trace e358c807049fd043 ]---
[... 68 lines cut ...]
I triyed call it in such way:
```
$ sudo ./scripts/patch-realsense-ubuntu-lts.sh build_usbcore_modules
```
in menuconfig menu I choose "Save", "OK", "Exit" and "Exit"
after build crashed with that text 
```
make: Leaving directory '/home/nvr/librealsense/ubuntu-xenial-master'
Compiling media/common modules
make: Entering directory '/home/nvr/librealsense/ubuntu-xenial-master'
scripts/Makefile.build:44: /home/nvr/librealsense/ubuntu-xenial-master/drivers/media/common/videobuf2/Makefile: No such file or directory
make[1]: *** No rule to make target '/home/nvr/librealsense/ubuntu-xenial-master/drivers/media/common/videobuf2/Makefile'.  Stop.
Makefile:1420: recipe for target '_module_/home/nvr/librealsense/ubuntu-xenial-master/drivers/media/common/videobuf2' failed
make: *** [_module_/home/nvr/librealsense/ubuntu-xenial-master/drivers/media/common/videobuf2] Error 2
make: Leaving directory '/home/nvr/librealsense/ubuntu-xenial-master'

```

Our kernel process looks like: 
- setup ubuntu on the virtual box 
- run remote build ( scripts/patch-realsense-ubuntu-lts.sh )
- copy drivers (uvcvideo.ko videodev.ko) as artifact (we dont replace accel and gypo because of non using it)
- replace them on the target system

Any thoughts about usbcore kernel errors?

MartyG-RealSense, 2022-11-28:
Hi @Allius27  The typical approach to installing the librealsense SDK from source code on a PC computer where you are using kernel patching would be to follow the instructions at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

This would involve first running the kernel patch script and then building librealsense from source code afterwards.  It is not usually necessary to add **build_usbcore_modules** to the kernel patch run instruction or copy and paste drivers.

*************

If your project only uses one camera per computer then a way of achieving a consistent installation on multiple computers may be to use CMake to build librealsense from source code with the **RSUSB backend** installation method, which bypasses the Linux kernel.  An RSUSB build of librealsense is therefore not dependent on Linux versions or kernel versions and does not require patching.  Instructions for an RSUSB install can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

Allius27, 2022-11-29:
I rebuild sdk with -DFORCE_RSUSB_BACKEND

And got this dmesg behavior 

<details>
  <summary>dmesg -w</summary>

```
[  108.275055] uvcvideo: Unable to create debugfs 2-3 directory.
[  108.275284] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input117
[  109.325649] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  109.329769] uvcvideo: Unable to create debugfs 2-3 directory.
[  109.329994] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input118
[  110.404196] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  110.406167] uvcvideo: Unable to create debugfs 2-3 directory.
[  110.406380] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input119
[  111.468938] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  111.472901] uvcvideo: Unable to create debugfs 2-3 directory.
[  111.473145] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input120

[  112.882242] usb 2-4: Process 1641 (atm-rsserver) called USBDEVFS_CLEAR_HALT for active endpoint 0x84
[  115.344551] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  115.347276] uvcvideo: Unable to create debugfs 2-3 directory.
[  115.347501] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input121
[  115.443983] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  115.446456] uvcvideo: Unable to create debugfs 2-3 directory.
[  115.446697] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input122
[  115.476630] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  115.477443] uvcvideo: Unable to create debugfs 2-3 directory.
[  115.586018] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  115.591399] uvcvideo: Unable to create debugfs 2-3 directory.
[  115.591713] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input123
[  115.714858] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  115.718846] uvcvideo: Unable to create debugfs 2-3 directory.
[  115.719093] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input124
[  115.822005] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  115.825739] uvcvideo: Unable to create debugfs 2-3 directory.
[  115.826101] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input125
[  115.982245] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  115.985392] uvcvideo: Unable to create debugfs 2-3 directory.
[  115.986181] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input126
[  116.123535] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  116.126466] uvcvideo: Unable to create debugfs 2-3 directory.
[  116.126766] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input127
[  116.271025] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  116.274563] uvcvideo: Unable to create debugfs 2-3 directory.
[  116.274928] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input128
[  116.422716] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  116.428516] uvcvideo: Unable to create debugfs 2-3 directory.
[  116.428916] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input129
[  116.590121] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  116.592903] uvcvideo: Unable to create debugfs 2-3 directory.
[  116.593956] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input130
[  116.736307] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  116.739743] uvcvideo: Unable to create debugfs 2-3 directory.
[  116.740181] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input131
[  116.888302] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  116.890565] uvcvideo: Unable to create debugfs 2-3 directory.
[  116.890825] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input132
[  117.038444] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  117.041492] uvcvideo: Unable to create debugfs 2-3 directory.
[  117.041860] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input133
[  117.192258] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  117.196573] uvcvideo: Unable to create debugfs 2-3 directory.
[  117.196844] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input134
[... cut ...]
[  284.769240] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input296
[  285.835424] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  285.840189] uvcvideo: Unable to create debugfs 2-3 directory.
[  285.840357] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input297
[  286.887818] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)

```
<details>

Allius27, 2022-11-29:
And after restoring original drivers I got the next 

command for restoring - *sudo apt install --reinstall linux-modules-extra-4.4.0-210-generic*

<details>
  <summary>dmesg -w</summary>

```
[   33.548207] usb 2-4: USB disconnect, device number 2
[   33.788609] usb 2-4: new SuperSpeed USB device number 3 using xhci_hcd
[   33.806057] usb 2-4: New USB device found, idVendor=8086, idProduct=0ad3
[   33.806067] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   33.806073] usb 2-4: Product: Intel(R) RealSense(TM) Depth Camera 415 
[   33.806079] usb 2-4: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415 
[   33.806084] usb 2-4: SerialNumber: 841113022293
[   33.807669] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   33.809435] uvcvideo: Unable to create debugfs 2-3 directory.
[   33.809601] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input31
[   33.810285] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   33.811002] uvcvideo: Unable to create debugfs 2-3 directory.
[   34.600233] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   34.603198] uvcvideo: Unable to create debugfs 2-3 directory.
[   34.603530] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input32
[   34.739958] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   34.744147] uvcvideo: Unable to create debugfs 2-3 directory.
[   34.744547] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input33
[   34.773781] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   34.775825] uvcvideo: Unable to create debugfs 2-3 directory.
[   34.844149] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   34.846997] uvcvideo: Unable to create debugfs 2-3 directory.
[   34.847220] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input34
[   34.884620] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   34.886637] uvcvideo: Unable to create debugfs 2-3 directory.
[   34.886989] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input35
[   34.982003] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   34.984126] uvcvideo: Unable to create debugfs 2-3 directory.
[   34.984393] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input36
[   35.137583] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   35.140379] uvcvideo: Unable to create debugfs 2-3 directory.
[   35.140627] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input37
[   35.284231] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   35.286366] uvcvideo: Unable to create debugfs 2-3 directory.
[   35.286603] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input38
[   35.419425] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   35.422427] uvcvideo: Unable to create debugfs 2-3 directory.
[   35.422743] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input39
[   35.572510] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   35.575978] uvcvideo: Unable to create debugfs 2-3 directory.
[   35.576556] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input40
[   35.727651] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   35.731338] uvcvideo: Unable to create debugfs 2-3 directory.
[   35.731596] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input41
[   35.895461] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   35.896995] uvcvideo: Unable to create debugfs 2-3 directory.
[   35.897199] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input42
[   36.050792] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   36.054252] uvcvideo: Unable to create debugfs 2-3 directory.
[   36.054490] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input43
[   36.215568] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   36.218263] uvcvideo: Unable to create debugfs 2-3 directory.
[   36.218472] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input44
[   36.362981] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   36.367958] uvcvideo: Unable to create debugfs 2-3 directory.
[   36.368343] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input45
[   36.525442] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   36.532434] uvcvideo: Unable to create debugfs 2-3 directory.
[... cut ...]
[   59.447273] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   59.448928] uvcvideo: Unable to create debugfs 2-3 directory.
[   59.449173] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input71
[   60.494320] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[   60.532660] uvcvideo: Unable to create debugfs 2-3 directory.
[   60.537106] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/input/input72

```
<details>

MartyG-RealSense, 2022-11-30:
If the linux modules were reinstalled with the command in the comment above then that would likely undo the changes to modules such as uvcvideo that were applied by the patch script.  So the patch script would need to be run again to re-patch the reinstalled modules.

Allius27, 2022-11-30:
wait a minute..  
if we build sdk with a key FORCE_RSUSB_BACKEND do we need  replace the uvcvideo and videodev kernel modules or we might keep original?

MartyG-RealSense, 2022-11-30:
When using RSUSB you do not need to apply patching to kernel modules, and doing so after the RSUSB installation has completed may break the RSUSB build.  You can keep the original modules.

Allius27, 2022-11-30:
I made some experiments with different realsense examples 

sdk version is 
```
git describe --contains HEAD
v2.50.0
```

we build it with keys 
```
mkdir build 
cd build
cmake \
	-DBUILD_EXAMPLES=ON \
	-DBUILD_GRAPHICAL_EXAMPLES=ON \
	-DBUILD_GLSL_EXTENSIONS=ON \
	-DCMAKE_BUILD_TYPE=Release \
	-DFORCE_RSUSB_BACKEND=ON \
	-DCMAKE_INSTALL_PREFIX=install \
	..
	
make install
```

go to install/bin directory 
```
cd install/bin 
```

LD_LIBRARY_PATH=../lib/ ./rs-color  
cause tons of  dmesg notes  (near 3-5 per second)
```
...
[16860.859481] uvcvideo 4-3:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[16860.859510] uvcvideo 4-3:1.2: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[16860.859620] usb 4-3: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[16860.861481] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/input/input160
...
```

LD_LIBRARY_PATH=../lib/ ./rs-capture 
only that for the all session
```
[16981.569071] uvcvideo 4-3:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[16981.569104] uvcvideo 4-3:1.2: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[16981.569162] usb 4-3: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[16981.571118] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/input/input172
[16981.762687] uvcvideo 4-3:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[16981.762724] uvcvideo 4-3:1.2: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[16981.762814] usb 4-3: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[16981.770397] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/input/input173
[16981.793848] uvcvideo 4-3:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[16981.793859] usb 4-3: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
```

LD_LIBRARY_PATH=../lib/ ./rs-hello-realsense
```
[17049.204847] uvcvideo 4-3:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[17049.205197] uvcvideo 4-3:1.2: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[17049.205287] usb 4-3: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[17049.209288] uvcvideo 4-3:1.2: Failed to set UVC probe control : -32 (exp. 48).
[17049.209719] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/input/input176
[17049.319451] uvcvideo 4-3:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[17049.319517] uvcvideo 4-3:1.2: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[17049.322307] usb 4-3: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[17049.324115] uvcvideo 4-3:1.2: Failed to set UVC probe control : -32 (exp. 48).
[17049.324456] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3:1.0/input/input177
[17049.346632] uvcvideo 4-3:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[17049.346646] usb 4-3: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
```

One note.
That is unavailable to operate sensors via rs2::syncer and FORCE_RSUSB_BACKEND. In our app by default running color sensor, but depth sensor via request. When we are turning it on application is catching error, but with changing drivers that works fine.

So.. the question is
Is it OK if the kernel catches errors when using the camera and SDK with FORCE_RSUSB_BACKEND compiling key?

MartyG-RealSense, 2022-11-30:
When changing options with an RSUSB build, there may be occasions when the request to change an option is not heard because RSUSB listens for changes less often than a kernel-patch based build of librealsense.  This is discussed at https://github.com/IntelRealSense/librealsense/issues/6952

At https://github.com/IntelRealSense/librealsense/issues/6921#issuecomment-668652875 advice is provided regarding editing the SDK's code to listen for device changes more often.  The modified SDK could then be built from its source code.

If you are satisfied with how RSUSB performs though then you can ignore the kernel errors.

Allius27, 2022-12-01:
>> https://github.com/IntelRealSense/librealsense/issues/6921#issuecomment-668652875 
unfortunately, that advice is acceptable for v2.36.0 release, not for v2.50.0, and not for latest master branch

and the second moment.
SDK build  with FORCE_RSUSB_BACKEND key 
When application is working, and frames are grabbing

that code works 30 seconds, instead of 15
thus 15 + 15 
```
rs2::frameset frames = pipe->wait_for_frames();
```

that 16 seconds 
thus 15 + 1 
```
rs2::frameset frames = pipe->wait_for_frames(1000);
```

and that 18 seconds 
thus 15 + 3
```
rs2::frameset frames = pipe->wait_for_frames(3000);
```

MartyG-RealSense, 2022-12-02:
If your script works correctly on your other computer - Gigabyte GB-BACE-3160 - with SDK 2.50.0 then that suggests that the script is okay and there is some hidden difference between the configuration of the Gigabyte computer and the computer that the video is looping on.

In a past case with two computers of the same model that seemed to have the same configuration, a script worked well on one of the computers but not the other.  It was later found that the computer that worked well had newer drivers installed than the computer that the script had problems on.  Updating the problematic computer's configuration to match the working computer resolved the problem.

Allius27, 2022-12-15:
I found the reason of additional 15 seconds of waiting frames 
the usb was recognized as 2.0 ( no idea why )

but when it recognizes as 3.0 there is  no troubles

thank for advice about -DFORCE_RSUSB_BACKEND compiling key

closed 2022-12-15 by Allius27

MartyG-RealSense, 2022-12-15:
It's excellent news that you found the source of your problem.  Thanks very much for the update!