realsenseai/librealsense #11157: Calculate object's volume with real sense D455 Java

GayanMDigital400, 2022-11-29:
Hi @MartyG-RealSense  @RealSenseCustomerSupport,

I want to measure the volume of objects by using the D455 Camera. Is there any example project developed using Java Language, or can you guys give guidance for this?

MartyG-RealSense, 2022-11-29:
Hi @GayanMDigital400  The best available editable project for this purpose is the RealSense SDK's Python example project **box_dimensioner_multicam.py**.  Despite its name, it can be used with a single RealSense camera.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

https://dev.intelrealsense.com/docs/box-measurement-and-multi-camera-calibration

There is not an equivalent of this project in Java, unfortunately.  The closest Java reference is one at https://github.com/IntelRealSense/librealsense/issues/9466 regarding measuring between two points on an image.

MartyG-RealSense, 2022-12-04:
Hi @GayanMDigital400  Do you require further assistance with this case, please?  Thanks!

GayanMDigital400, 2022-12-05:
Hi @MartyG-RealSense, while working with the python example you gave, there are problems with object detection. It can't detect an actual object that I have placed on the chessboard. 

[image]
[image]

I hope you understood me. I wait for your helping. Thanks a lot.

MartyG-RealSense, 2022-12-05:
The box_dimensioner_multicam.py example is sensitive to distance.  If the camera is too far or too close to the chessboard image then the green box may be rendered incorrectly.  This is discussed at https://github.com/IntelRealSense/librealsense/issues/10100

The chessboard is also usually placed flat on the floor instead of mounted on a wall.

closed 2022-12-20 by MartyG-RealSense