realsenseai/librealsense #11171: Hello-World xcopy error

KirkPen, 2022-12-05:
| Camera Model                       | {D435i} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC
| SDK Version                            |           2.51.1.4348               | 
| Language                            |  C++                          | 
| Segment			|  {Robot                        | 

### Issue Description

Error	MSB3073	The command "xcopy /y "C:\Program Files\Intel RealSense SDK 2.0\bin\x64\realsense2.dll" "C:\Program Files (x86)\Intel RealSense SDK 2.0\samples\x64\Debug\"
:VCEnd" exited with code 4.	hello-realsense	C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets	159

MartyG-RealSense, 2022-12-06:
Hi @KirkPen  Windows has two program directories, 'Program Files' and 'Program Files (x86)'  The RealSense SDK 2.0 installs to Program Files (x86) on a Windows computer.  The **realsense2.dll** file is located in the Program Files (x86) directory, whereas the xcopy command above looks for it in the **Program Files** folder.

`C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64\realsense2.dll`

MartyG-RealSense, 2022-12-06:
I will highlight this issue to my Intel RealSense colleagues.  Thanks very much for your patience!

RealSenseCustomerSupport, 2022-12-08:
----------------------------------------------

Please check the path in the project property page -> Build-Events -> Post-Built event. If the path contains macro variable (i.e., librealsenseSDK), please check path of the variable as well. Thank you.

Attachment(s):
xcopy_path.png - https://realsensesupport.intel.com/attachments/token/fFd916WHKxhUIUnt36t04ylf2/?name=xcopy_path.png

KirkPen, 2022-12-08:
Ya thats what I am trying to tell you, your macro is wrong, it is looking at c:\program files for the file to copy, and it should be looking at the c:\program filesx86. Please fix the macro as I can't edit it. My fix is to copy the files and end up with duplicates so the build will work. 
[image]
[image]

RealSenseCustomerSupport, 2022-12-08:
----------------------------------------------

The librealsenseSDK path is based on the ProgramFiles path and it looks like VC2022 has changed the ProgramFiles directory from "Program Files (x86)" to "Program Files"

The macro is defined in the intel.realsense.prop file under Intel RealSense SDK2.0 folder.
Please open the file with elevated permission and change:
`$(ProgramFiles)\Intel RealSense SDK 2.0`
to
`$(MSBuildProgramFiles32)\Intel RealSense SDK 2.0`

After that, the project should be built correctly on vc2022.

Attachment(s):
vc2022_prgramfiles_path.jpg - https://realsensesupport.intel.com/attachments/token/QQcNK2TK35rZxuoTHP7yarzDt/?name=vc2022_prgramfiles_path.jpg

KirkPen, 2022-12-08:
Thank you, that did it. I did spend some time trying to make the enviorment variable for programfiles (x86) to work but it did not like it.

MartyG-RealSense, 2022-12-08:
@KirkPen  It's great to hear that you had success!

@RealSenseCustomerSupport  Thanks so much for helping @KirkPen

KirkPen, 2022-12-14:
> Hi @KirkPen <https://github.com/KirkPen> Do you require further
> assistance with this case, please? Thanks!

No i edit the file. Can you get them to update to fix or put a note in
about it?

MartyG-RealSense, 2022-12-14:
I have passed your suggestion to update the intel.realsense.prop file to @RealSenseCustomerSupport - thanks again!

MartyG-RealSense, 2022-12-15:
A proposal has been submitted to the RealSense developer team to provide compatibility with VS2022 in a future SDK release, as described at https://github.com/hsuys/librealsense/commit/51e41b99485edf1f1613b6641b48866dbf5dace3

MartyG-RealSense, 2022-12-22:
Update: the change has been merged into the RealSense SDK so that it will be in the next SDK release, as described at https://github.com/IntelRealSense/librealsense/pull/11217

MartyG-RealSense, 2022-12-28:
Case closed due to the proposed change having been merged into the RealSense SDK to be available in an upcoming release.

closed 2022-12-28 by MartyG-RealSense