realsenseai/librealsense #11172: Unable to compile custom cpp applications using realsense header files.

VishnuRamTK, 2022-12-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux built from petalinux 2020.2 tool  | 
| Kernel Version (Linux Only)    |  (5.4.0)                                         | 
| Platform                                 | Kria KV260 FPGA  |
| SDK Version                            |  { legacy / 2.51.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am trying to use librealsense header file rs.hpp in my custom c++ application for some image processing tasks. However when i compile a simple rs-hello-realsense.cpp file that already comes with the sdk examples i get an error -
**collect2: error: ld returned 1 exit status**

MartyG-RealSense, 2022-12-06:
Hi @VishnuRamTK  The RealSense SDK example programs are built in a way that may cause librealsense libraries to be missing if those programs' code is copied into a user-created project.

The starter project at the link below has the same script as rs-hello-realsense but a more complex CMakeLists.txt file that enables the librealsense libraries to be found by a standalone project.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

VishnuRamTK, 2022-12-07:
Hi @MartyG-RealSense ,
I tried the example in the above github page and it works. However i want to use the RGB camera stream from Intel 435i and feed it to program in https://www.hackster.io/code_files/574226/download. This code is obtained following the link below -
https://www.hackster.io/dsp2/easy-machine-learning-on-ubuntu-with-the-xilinx-kria-kv260-c77ed1

I included the rs.hpp header in to the same program however i could not compile it. The way the above program is compiled adds to the complication. The snippet of how the program is compiled below(also taken from the above link) -

g++ -std=c++17 -O3 main.cpp -o person_detection \
   -I/usr/include/opencv4 \
   -lpthread \
   -lopencv_core \
   -lopencv_video \
   -lopencv_videoio \
   -lopencv_imgproc \
   -lopencv_imgcodecs \
   -lopencv_highgui \
   -lvitis_ai_library-refinedet

I even moved the librealsense2 folder to /usr/include folder where the compiler searches for the libraries and also added a flag -llibrealsense2_rs .However i keep getting ld error. I am quite not sure how to modify the CMakelists.txt to look for all the above libraries and realsense2_Library while compiling.

[image]

MartyG-RealSense, 2022-12-07:
There was a case where a RealSense user experienced the same set of errors and a RealSense team member provided them with advice for a solution at https://github.com/IntelRealSense/librealsense/issues/5651#issuecomment-576017475

VishnuRamTK, 2022-12-08:
Thanks for the reply @MartyG-RealSense. I could somehow modify the CMakelist.txt file and tried including all the dependent libraries and could compile the code succesfully. Thanks for all the references. 

Modifies CMakelist.txt file below for future reference -

cmake_minimum_required(VERSION 3.1.0)

project(Realsensepersondetector)

# Save the command line compile commands in the build output
set(CMAKE_EXPORT_COMPILE_COMMANDS 1)

set(ADD_INCLUDE_DIRS "/usr/include/opencv2")
# Make project require C++17
include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++17" COMPILER_SUPPORTS_CXX17)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if(COMPILER_SUPPORTS_CXX17)
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c17")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++17")
elseif(COMPILER_SUPPORTS_CXX0X)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
endif()
# Simple non robust way to find the librealsense library
if(WIN32)
    if("${CMAKE_SIZEOF_VOID_P}" STREQUAL "8")
        set(LIBRARY_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\lib\\x64")          # TODO: Update this variable to correct path - folder where realsense2.lib is found
        set(DLL_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\bin\\x64")              # TODO: Update this variable to correct path - folder where realsense2.dll is found
    else()
        set(LIBRARY_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\lib\\x86")          # TODO: Update this variable to correct path - folder where realsense2.lib is found
        set(DLL_DIR "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\bin\\x86")              # TODO: Update this variable to correct path - folder where realsense2.dll is found
    endif()
    set(PROJECT_BINARY_OUTPUT_PATH "${PROJECT_BINARY_DIR}/\$\(Configuration\)")                # TODO: Update this variable to correct path - folder to which your project will compile
    set(ADDITIONAL_INCLUDE_DIRS "C:\\Program Files (x86)\\Intel RealSense SDK 2.0\\include")   # TODO: Update this variable to correct path - folder where librealsense2 folder is found
endif()

find_library(REALSENSE2_FOUND realsense2 HINTS ${LIBRARY_DIR} REQUIRED)
if(NOT REALSENSE2_FOUND)
    SET(REALSENSE2_FOUND "realsense2")
    message(WARN "Failed to find_library(realsense2)")
endif()
# getting-started
add_executable(rs_person_detection main.cpp)
include_directories(rs_person_detection ${ADD_INCLUDE_DIRS} ${ADDITIONAL_INCLUDE_DIRS})
target_link_libraries(rs_person_detection ${REALSENSE2_FOUND} pthread opencv_core opencv_video opencv_videoio opencv_imgproc opencv_imgcodecs opencv_highgui vitis_ai_library-refinedet)
set_target_properties (rs_person_detection PROPERTIES FOLDER Examples)

# Post Build script to copy realsense2.dll
if(WIN32)
message(STATUS "Adding Post build script to copy realsense2.dll to project's binary folder")
message(STATUS "Will try to copy from ${DLL_DIR} to ${PROJECT_BINARY_OUTPUT_PATH}")
add_custom_command(TARGET rs_person_detection POST_BUILD        # Adds a post-build event to rs-getting-started
    COMMAND ${CMAKE_COMMAND} -E copy_if_different              # which executes "cmake - E copy_if_different..."
        "${DLL_DIR}/realsense2.dll"                            # <--this is in-file
        ${PROJECT_BINARY_OUTPUT_PATH})                 # <--this is out-file path
endif()

install(
    TARGETS

    rs_person_detection

    RUNTIME DESTINATION
    ${CMAKE_INSTALL_PREFIX}/bin
)

closed 2022-12-08 by VishnuRamTK

MartyG-RealSense, 2022-12-08:
It's great to hear that you achieved a solution, @VishnuRamTK - thanks so much for sharing your solution with the RealSense community!