realsenseai/librealsense #11178: Save and load RGB and Depth Images for further pyrealsense manipulation

marcoluis97, 2022-12-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.15.0-56-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0 | 
| Language                            |   python   | 
| Segment			|  Robot   | 

### Issue Description

Greetings! 

I am looking to save depth and rgb frames from my Intel RealSense D455 at my will given a certain command. After that I wish to retrieve that same information to further process it and get distances of pixels present in the frame in a separate script from the one that saves the frames. 
From what I understand the functions from the python wrapper that allow to get distances of pixels and deproject pixels to 3d points receive as input the frame in a pyrealsense type (more specifically <class 'pyrealsense2.pyrealsense2.video_frame'>
<class 'pyrealsense2.pyrealsense2.depth_frame'> types for color/rbg and depth frames respectively). I have searched so far through past issues posted but have not found a way to save this type of variable.

MartyG-RealSense, 2022-12-07:
Hi @marcoluis97  Instead of having a separate script for getting pixel distance, an alternative possibility may be to use Python **multithreading** so that the capture and processing is handled separately from the streaming.  An example of such a script can be found at the link below, where one thread had the camera live-stream and the other thread recorded captured data to video.

https://support.intelrealsense.com/hc/en-us/community/posts/360033158094-Recording-Video-in-60fps-with-threading-and-queue-Python-

Another way may be to use the RealSense SDK's **Keep()** instruction to store frames in the computer's memory.  Once the pipeline is closed then you can then perform batch-processing on the stored frames all at the same time, such as applying alignment and post-processing, and then save the frames to file.

Because Keep() stores frames in memory though, it can quickly consume the computer's available memory.  A low-end computing device may be able to store 10 seconds worth of frames, whilst a computer such as a PC with higher memory capacity may be able to store 30 seconds of frames.  It may be possible to extend this storage duration by releasing frames to free up memory.

In regard to converting a numpy array back to an SDK format such as rs2_frame, as far as I am aware this remains an unsolved problem unfortunately.

MartyG-RealSense, 2022-12-16:
Hi @marcoluis97  Do you require further assistance with this case, please?  Thanks!

marcoluis97, 2022-12-19:
Hi @MartyG-RealSense! Thanks for the suggestions. I've already fixed my issue and require no futher assistance. Thank you!

closed 2022-12-19 by marcoluis97

MartyG-RealSense, 2022-12-19:
That's excellent to hear, @marcoluis97 - thanks very much for the update!

mirzafahad, 2023-02-18:
Hi @marcoluis97,

Were you able to save the frames the way you intended?