realsenseai/librealsense #11181: Ros2 foxy no point cloud and not all topics

zachrizzo, 2022-12-06:
when I run the point cloud demo (or any demo)  not all topics are shown and I never see a point cloud in rviz
I also have not ben able to get and any topics involving the IMU to show up in rviz

MartyG-RealSense, 2022-12-07:
Hi @zachrizzo  Are you using the **4.51.1** ROS2 wrapper version, please?  If you are, does the pointcloud publish if you use the launch instruction below?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true`

In regard to the IMU topics, they are disabled by default in the wrapper.  They can be published by adding the commands below to the end of the launch instruction:

`enable_gyro:=true enable_accel:=true`

If you wish to publish the IMU topics as a single combined **/imu** topic instead of two separate accel and gyro topics, add this:

`unite_imu_method:=2`

So a launch instruction using all these commands would look like this:

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true enable_gyro:=true enable_accel:=true unite_imu_method:=2`

zachrizzo, 2022-12-08:
I believe I have 4.54.1. I tried both those and this is the error i get for them

when i run the first one this is the out put

`[INFO] [launch]: All log files can be found below /home/zach/.ros/log/2022-12-07-21-06-44-917179-ubuntu-222705
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [222707]
[realsense2_camera_node-1] [INFO] [1670472405.658843196] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1670472405.659159453] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1670472405.659269342] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1670472405.719145935] [camera.camera]: Device with serial number 213522071748 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1670472405.719314800] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1670472405.719411024] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1670472405.720463189] [camera.camera]: Device with port number 2-3.1 was found.
[realsense2_camera_node-1] [INFO] [1670472405.720546837] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1670472406.011946119] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1670472406.014058577] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1670472406.014213713] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1670472406.014286354] [camera.camera]: Device Serial No: 213522071748
[realsense2_camera_node-1] [INFO] [1670472406.014344658] [camera.camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1670472406.014400082] [camera.camera]: Device FW version: 05.13.00.55
[realsense2_camera_node-1] [INFO] [1670472406.014501139] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1670472406.014563315] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [WARN] [1670472406.089143015] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1670472406.104072459] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1670472406.126630738] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1670472406.129566176] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1670472406.129925313] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1670472406.173757897] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1670472406.203239216] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1670472406.266622353] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1670472406.267230772] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1670472406.276137052] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1670472406.286492140] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1670472406.288421620] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1670472406.345505209] [camera.camera]: 

`

when i run the second one I get 

`[INFO] [launch]: All log files can be found below /home/zach/.ros/log/2022-12-07-21-09-36-929643-ubuntu-222781
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [222783]
[realsense2_camera_node-1] [INFO] [1670472577.849710962] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [1670472577.850158484] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1670472577.850247476] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [1670472577.910909061] [camera.camera]: Device with serial number 213522071748 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1670472577.911093030] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1670472577.911155302] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1670472577.912177483] [camera.camera]: Device with port number 2-3.1 was found.
[realsense2_camera_node-1] [INFO] [1670472577.912255915] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1670472577.917012641] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1670472577.918459879] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1670472577.918581416] [camera.camera]: Device Name: Intel RealSense D435I
[... cut ...]
[realsense2_camera_node-1] [WARN] [1670472580.076421056] [camera.camera]: Hardware Notification:Depth stream start failure,1.67047e+12,Error,Hardware Error

`
how ever i do not see the topics in rviz and i only see the camera topic and see the depth and rgb image.

when i installed the ros wrapper i wasnt able to run the rodep comand so i just colcon build.( this is because i got the error :
Cannot locate rosdep definition for [pyrealsense2]

i figure it was fine because i was able to see the depth image topic still. do you think this could be the issue?

zachrizzo, 2022-12-08:
this is the thread i used to install librealsense
https://github.com/IntelRealSense/librealsense/issues/10416

this are the commands I'm used to install  

sudo apt-get update && sudo apt-get -y upgrade
sudo apt-get install -y --no-install-recommends \
    python3 \
    python3-setuptools \
    python3-pip \
    python3-dev

sudo apt-get install -y git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev

sudo apt-get install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev

sudo apt-get install libusb-1.0-0-dev

sudo apt-get install python3.9-dev

git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense/
mkdir build && cd build
cmake ../ -DFORCE_RSUSB_BACKEND=false -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
sudo make uninstall && make clean && make && sudo make install

I'm on jetpack 5 on a jetson Xavier NX , which is ubuntu 20.04 

when i run the realsense view i am able to run everything fine except I get one warring which is there is "there is no IMU calibration" 

but i can see it working in the 2D view.

I had to install it this way because I see you guys don officially support jetpack 5.0, and don't plan to. ( at least that is what i have seen in another thread). I don't understand why you guys don't support jetpack 5.0 because that's the only jetpack that can run ros2.

MartyG-RealSense, 2022-12-09:
Whilst there is not official support for JetPack 5 at the time of writing this, it can work unofficially.  Typically the main problem that is reported is with the IMU whilst depth and color function normally on JetPack 5.  

I note from your earlier comment at https://github.com/IntelRealSense/librealsense/issues/10416#issuecomment-1334951728 that you were able to get the IMU working with librealsense SDK applications such as realsense-viewer by launching in sudo mode.  But you are still having IMU compatibility problems with the RealSense ROS wrapper?

I will query my Intel RealSense colleagues about official JetPack 5 support and the current IMU problem.

zachrizzo, 2022-12-13:
Thank you so much for looking into this, I am looking forward to the release

MartyG-RealSense, 2022-12-19:
Hi @zachrizzo  Do you require further assistance with this case, please? Thanks!

ryanmaxwell96, 2022-12-19:
I really hope to get Jetpack 5 support soon because I don't really want to have to revert back to Jetpack 4.6 because it means needing a computer with ubuntu 18.04 and I already upgraded mine to Ubuntu 20.04

MartyG-RealSense, 2022-12-19:
My understanding is that it would be introduced sooner rather than later after the New Year 2023, though I do not have a precise time estimate to provide at the time of writing this.

ryanmaxwell96, 2022-12-19:
That would be excellent, thanks for the info!

zachrizzo, 2022-12-22:
Yes very excited!

MartyG-RealSense, 2023-01-18:
Update: support for JetPack **5.0.2** has now been merged into librealsense for inclusion in the next SDK release.

zachrizzo, 2023-01-19:
@MartyG-RealSense  Amazing! thank you so much! Do I need to uninstall all or can I just install the latest? Also, does this update include ros2?

MartyG-RealSense, 2023-01-19:
For your reference, the Pull Request that added the support is at https://github.com/IntelRealSense/librealsense/pull/11242

A merge means that a change has been incorporated into the **development** version of librealsense, the next version of the SDK that is currently being developed.  When a development version is released, it becomes the latest **master** version.

The development branch of the SDK can be built from source code at the link below.

https://github.com/IntelRealSense/librealsense/tree/development

When JetPack support is updated in librealsense, the change automatically applies to the RealSense ROS wrapper too.

ryanmaxwell96, 2023-01-19:
Awesome!

MartyG-RealSense, 2023-07-18:
Case closed due to JetPack 5.0.2 support now available.

closed 2023-07-18 by MartyG-RealSense