realsenseai/librealsense #11182: Access Violation message

noodle875, 2022-12-07:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2022-12-07:
Hi @noodle875  Please first check whether permission to access the camera hardware and permission for applications to access the camera are both set to 'On' in the Windows camera privacy settings.  This can be done using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

If you find that both settings are already On then please try completely uninstalling and reinstalling the two RealSense camera drivers (Depth and RGB) in the Windows **Device Manager** interface using the instructions at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

In Step 4 of the instructions, if there is not a tick-box to select then skip to Step 5.

noodle875, 2022-12-14:
Oh I had forgotten about this.

closed 2022-12-14 by noodle875