realsenseai/librealsense #11185: Two RGB stream recording

Aki1608, 2022-12-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.14.00.00) | 
| Operating System & Version |   {Ubuntu 20.04} | 
| Kernel Version (Linux Only)    |  (5.10.104-tegra)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  { legacy / 2.53.1 }                          | 
| Language                            |  {N/A}                          | 
| Segment			|  {NA }                          | 

### Issue Description
This is more like a question than a issue.

I want to know if it is possible to record video streams from both RGB camers?

Aki1991, 2022-12-08:
Ah Okay. Thank you for the quick reply. :)

I am closing this.

Aki1608, 2022-12-08:
Ah okay. Understood. Thank you so much for the quick reply. :)

I am closing this.

Aki1608, 2022-12-08:
I have another question. Is it possible to record a video of RGB frame and point cloud from SDK?

MartyG-RealSense, 2022-12-08:
You are very welcome, @Aki1608    :)

You could not record a point cloud, but what you could do is record a **bag** file of the RGB and depth streams and then use that bag file as the data source in a script instead of a live camera and generate a pointcloud from the stored individual stream data in real-time.

Individual depth and RGB streams can also be saved as non-interactive **.avi** video files, like in the Python script for doing so at https://github.com/IntelRealSense/librealsense/issues/2731#issuecomment-529938267

Aki1608, 2022-12-09:
Okay.  Thank you. :)

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense