realsenseai/librealsense #11187: Depth Colorization Algorithm Wrong

AlexanderKhazatsky, 2022-12-08:
Intel has posted the following algorithm for depth compression: https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras

After a quick scan, this seems to have the best recovered PSNR to algorithm complexity ratio, so it seems extremely important to make sure this post is usable.

I've found two critical issues with the formula presented (attached below). First, there are clear counterexamples that do not result in an invertible transformation. In fact, I'm actually not sure when it DOES work. So I assume there is a critical issue in the posted formula. Consider the case where:
depth = 5
d_min = 1
d_max = 100

This gets mapped to the RGB value (255, 61, 61), which then gets mapped back to the depth value of 0 instead of 5.

The paper claims that the encoder is implemented in the librealsense SDK, but I am not able to find it. Could somebody please try and find the discrepancy between the blog post and the presumably correct SDK implementation?

MartyG-RealSense, 2022-12-08:
Hi @AlexanderKhazatsky  The depth image compression by colorization white-paper document was a proof-of-concept experiment published some years ago and whilst it is possible to implement the compression, it is now recommended that it not be used and alternative depth compression methods be used instead, such as **LZ4**, as discussed at https://github.com/IntelRealSense/librealsense/issues/8117

If your preference is to continue with depth compression by colorization then there are limited references available for debugging problems with it unfortunately due to the experimental nature of the paper and the small number of RealSense users who attempted to use it.  The best available reference is at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4419463222931-Depth-Image-Colorization-Formula

AlexanderKhazatsky, 2022-12-08:
Thanks for the references!! It seems that LZ4 is a lossless algorithm that operates on single frames. I am reading from 6 depth cameras and will have thousands of hours of data, so I will likely need more aggressive compression. Do you have suggestions for video based lossy algorithms?

MartyG-RealSense, 2022-12-09:
Instead of using compression, setting the frames per second (FPS) speed to 6 FPS (or 5 FPS on the D455 camera model) instead of the typical 30 FPS should provide a subtantial reduction in data size.

MartyG-RealSense, 2022-12-30:
Case closed due to no further comments received.

closed 2022-12-30 by MartyG-RealSense

cheind, 2024-10-17:
Although this comes late in the game, for future reference, here is a vectorized Python implementation with the equations corrected 

https://github.com/cheind/hue-depth-encoding