realsenseai/librealsense #11189: [MSVC][std:c++latest] Librealsense failed with error C2666: 'librealsense::platform::backend_device_group::operator ==': overloaded functions have similar conversions

Zhaojun-Liu, 2022-12-08:
**Description:**
After the MSVC team implemented [P2468R2 The Equality Operator You Are Looking For](https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2022/p2468r2.html), we got compiler errors with /std:c++latest like below. The way this fails is that the equality operator here https://github.com/IntelRealSense/librealsense/blob/master/src/backend.h#L538 and https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_frame.hpp#L79 are missing 'const'. Could you please look this issue? Thanks.

**Note:** this issue will be reproduced on next release version of VS(17.6 or later).

**Repro steps:**
1. open VS2019 x64 tools command (Or VS2022)
2. git clone https://github.com/IntelRealSense/librealsense F:\gitP\IntelRealSense\librealsense
3. mkdir F:\gitP\IntelRealSense\librealsense\build_amd64
4. cd F:\gitP\IntelRealSense\librealsense\build_amd64
5. cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.18362.0 -DCMAKE_BUILD_TYPE=Release -DBUILD_LEGACY_LIVE_TEST=true -DBUILD_EXAMPLES=true -DBUILD_WITH_TM2=true -DCHECK_FOR_UPDATES=true .. 
6. set _CL_=/std:c++latest /D_HAS_AUTO_PTR_ETC=1 /D_HAS_DEPRECATED_ALLOCATOR_MEMBERS /D_SILENCE_CXX17_ITERATOR_BASE_CLASS_DEPRECATION_WARNING /Zc:strictStrings- /Zc:char8_t-
7. msbuild /m /p:Platform=x64 /p:Configuration=Release librealsense2.sln /t:Rebuild

**Error info:**
```
ds5-factory.cpp
F:\gitP\IntelRealSense\librealsense\src\context.h(57,41): error C2666: 'librealsense::platform::backend_device_group::operator ==': overloaded functions have similar conversions
F:\gitP\IntelRealSense\librealsense\src\backend.h(538,18): note: could be 'bool librealsense::platform::backend_device_group::operator ==(const librealsense::platform::backend_device_group &)'
F:\gitP\IntelRealSense\librealsense\src\types.h(1057,17): note: or       'bool librealsense::operator ==(const librealsense::hid_device_info &,const librealsense::hid_device_info &)'
F:\gitP\IntelRealSense\librealsense\src\types.h(1026,17): note: or       'bool librealsense::operator ==(const librealsense::usb_device_info &,const librealsense::usb_device_info &)'
F:\gitP\IntelRealSense\librealsense\src\types.h(993,17): note: or       'bool librealsense::operator ==(const librealsense::uvc_device_info &,const librealsense::uvc_device_info &)'
F:\gitP\IntelRealSense\librealsense\src\types.h(963,17): note: or       'bool librealsense::operator ==(const rs2_intrinsics &,const rs2_intrinsics &)'
F:\gitP\IntelRealSense\librealsense\src\types.h(561,17): note: or       'bool librealsense::operator ==(const librealsense::stream_profile &,const librealsense::stream_profile &)'
F:\gitP\IntelRealSense\librealsense\src\types.h(471,17): note: or       'bool librealsense::operator ==(const librealsense::pose &,const librealsense::pose &)'
```
```
unit-tests-post-processing.cpp
F:\gitP\IntelRealSense\librealsense\unit-tests\unit-tests-post-processing.cpp(653,13): error C2666: 'rs2::stream_profile::operator ==': overloaded functions have similar conversions
F:\gitP\IntelRealSense\librealsense\include\librealsense2\hpp\rs_frame.hpp(79,14): note: could be 'bool rs2::stream_profile::operator ==(const rs2::stream_profile &)'
F:\gitP\IntelRealSense\librealsense\unit-tests\../include/librealsense2/rs_advanced_mode.hpp(380,13): note: or       'bool operator ==(const STAFactor &,const STAFactor &)'
F:\gitP\IntelRealSense\librealsense\unit-tests\../include/librealsense2/rs_advanced_mode.hpp(374,13): note: or       'bool operator ==(const STCensusRadius &,const STCensusRadius &)'
F:\gitP\IntelRealSense\librealsense\unit-tests\../include/librealsense2/rs_advanced_mode.hpp(365,13): note: or       'bool operator ==(const STDepthTableControl &,const STDepthTableControl &)'
F:\gitP\IntelRealSense\librealsense\unit-tests\../include/librealsense2/rs_advanced_mode.hpp(360,13): note: or       'bool operator ==(const STAEControl &,const STAEControl &)'
F:\gitP\IntelRealSense\librealsense\unit-tests\../include/librealsense2/rs_advanced_mode.hpp(344,13): note: or       'bool operator ==(const STColorCorrection &,const STColorCorrection &)'
F:\gitP\IntelRealSense\librealsense\unit-tests\../include/librealsense2/rs_advanced_mode.hpp(337,13): note: or       'bool operator ==(const STHdad &,const STHdad &)'
F:\gitP\IntelRealSense\librealsense\unit-tests\../include/librealsense2/rs_advanced_mode.hpp(327,13): note: or       'bool operator ==(const STSloPenaltyControl &,const STSloPenaltyControl &)'
```

**Detailed log:** 
[Librealsense_build.log](https://github.com/IntelRealSense/librealsense/files/10182840/Librealsense_build.log)

Zhaojun-Liu, 2022-12-09:
Hi @MartyG-RealSense, this issue is indeed related to the c++latest. May I understand it to be that Librealsense only supports c++11 now, not c++latest?

MartyG-RealSense, 2022-12-10:
With further information provided at https://github.com/IntelRealSense/librealsense/issues/10933#issuecomment-1343970151 the problem was able to be reproduced by my colleagues.  I will keep you updated when I have more information to provide.  Thanks again!

MartyG-RealSense, 2022-12-22:
After my Intel RealSense colleagues discussed this issue of C++latest, they concluded that the errors below should be quick to fix.

```
C:\Intel\src\librs\librealsense\src\media\record\record_sensor.cpp(216,97): error C2385: ambiguous access of 'recordable' [C:\Intel\src\librs\librealsense\build\realsense2.vcxproj]
         
C:\Intel\src\librs\librealsense\src\mf\mf-uvc.cpp(1133,24): error C2440: 'initializing': cannot convert from '_InIt' to '_InIt &' [C:\Intel\src\librs\librealsense\build\realsense2.vcxproj]
```

This is because it is already restricted in step 7 of the list of repro instructions.

`msbuild /m /p:Platform=x64 /p:Configuration=Release librealsense2.sln /t:Rebuild`

Without restriction, i.e if simply set **CL=/std:c++latest**, there are 52 errors and the constant need to maintain the latest C++ support if the SDK were to claim support for c++latest.  It was concluded that supporting c++latest in general would require a large effort, as today's 'latest' is not tomorrow's latest.

MartyG-RealSense, 2022-12-28:
Hi @Zhaojun-Liu  Do you require further assistance with this case, please?  Thanks!

Zhaojun-Liu, 2022-12-29:
Hi @MartyG-RealSense,
You mean this issue is not currently going to be fixed, right? Or, I can further understand that Librealsense  does not plan to support c++latest for the time being, right?

MartyG-RealSense, 2022-12-29:
You are correct, it is not planned at this time that c++latest is supported due to the large effort involved in keeping the support updated.

Zhaojun-Liu, 2022-12-29:
> You are correct, it is not planned at this time that c++latest is supported due to the large effort that would be involved in keeping the support updated.

Ok, I got it. Thank you. Thank you for the reply.

MartyG-RealSense, 2023-01-03:
Hi @Zhaojun-Liu  Bearing in mind that c++latest support is not planned, do you require further assistance with this case please?  Thanks!

Zhaojun-Liu, 2023-01-03:
@MartyG-RealSense I have no further require with this case. You can close it now.  Thank you.

MartyG-RealSense, 2023-01-03:
Thanks very much @Zhaojun-Liu for the update.

closed 2023-01-03 by MartyG-RealSense