realsenseai/librealsense #11195: Infrared camera & OpenCV

Vermilion016, 2022-12-09:
Excuse me.
I am learning c++ and opencv.
[https://dev.intelrealsense.com/docs/rs-dnn](url)
I'm thinking about how to apply Color Stream to Infrared stream with the function of the above URL.
Is there a better way?

MartyG-RealSense, 2022-12-09:
Hi @Vermilion016  If you are seeking to align color to infrared, usually this is not possible with conventional alignment methods because one of the two stream types has to be Depth.  There are workarounds though.  At https://github.com/IntelRealSense/librealsense/issues/5963#issuecomment-638403524 a RealSense user shared their C++ / OpenCV script for aligning left and right infrared and color.

Vermilion016, 2022-12-10:
OK. I will try.
Thank you very much.

closed 2022-12-10 by Vermilion016