realsenseai/librealsense #11196: L515 not detected by realsense viewer.

ayusefi, 2022-12-09:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

MartyG-RealSense, 2022-12-09:
Hi @ayusefi  Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end). 
 Does doing so make any difference?

ayusefi, 2022-12-09:
I have a D415 stereo camera it works fine. But the L515 is not being detected. The PC detects the USB but the realsense_viewer 
does not. I use Windows 11.

MartyG-RealSense, 2022-12-09:
Can you confirm if you have tried reversing the L515's USB connector, please?  This reversal solution is for L515 only and does not affect 400 Series cameras such as L515.

closed 2022-12-30 by MartyG-RealSense