realsenseai/librealsense #11197: Error in box_dimensioner_multicam_demo.py in chessboard calibration?

Ankush1909IIT, 2022-12-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Trying to use box_dimesioner_multicademo.py. It is first calibrating the chessboard then telling to place the chessboard again. This s the error comment 
RMS error for calibration with device number 151322067505 is : 0.004997414131450382 m
Place the chessboard on the plane where the object needs to be detected..

It should ask to place the object.

MartyG-RealSense, 2022-12-10:
Hi @Ankush1909IIT  The **box_dimensioner_multicam.py** example program is sensitive to the distance of the camera from the chessboard.  There will be a 'sweet spot' that is not too close to the board and not too far.  I would suggest trying a distance that is not less than 0.5 m and not more than 1 m if possible.

MartyG-RealSense, 2022-12-17:
Hi @Ankush1909IIT  Do you require further assistance with this case, please?  Thanks!

Ankush1909IIT, 2022-12-20:
Yes, I have been able to get the detection using the sweet spot. Thankyou.

MartyG-RealSense, 2022-12-20:
That's great to hear.  Thanks very much for the update!

closed 2022-12-28 by MartyG-RealSense