realsenseai/librealsense #11206: Building RealSenseBagReader.cpp failed

viisautta, 2022-12-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | 5.14.0.0 | 
| Operating System & Version |  Feodra 37  | 
| Kernel Version (Linux Only)    | 6.0.1| 
| Platform                                 | PC |
| SDK Version                            | 5.23.1 | 
| Language                            |  c++ | 
| Segment			| others | 

 ### Issue Description

After Open3D was built, modified some configs as below:
...
BUILD_OPEN3D_EXAMPLES            ON
...
Open3D_DIR                                      /usr/local/open3d/lib64/cmake/Open3D
...

[o@192 build]$ make
[  3%] Built target libcurl
[  4%] Built target fw
[ 10%] Built target realsense-file
[ 46%] Built target realsense2
[ 46%] Built target rs-imshow
[ 47%] Built target rs-grabcuts
[ 47%] Built target rs-latency-tool
[ 47%] Built target rs-dnn
[ 48%] Built target rs-depth-filter
[ 49%] Built target rs-rotate-pc
[ 50%] Built target rs-kinfu
[ 50%] Linking CXX executable RealSenseBagReader
/usr/bin/ld: CMakeFiles/RealSenseBagReader.dir/cpp/RealSenseBagReader.cpp.o: in function `main':
/home/o/Public/librealsense-2.53.1/wrappers/open3d/cpp/RealSenseBagReader.cpp:89: undefined reference to `open3d::t::io::RSBagReader::RSBagReader(unsigned long)'
/usr/bin/ld: /home/o/Public/librealsense-2.53.1/wrappers/open3d/cpp/RealSenseBagReader.cpp:90: undefined reference to `open3d::t::io::RSBagReader::Open(std::string const&)'
/usr/bin/ld: /home/o/Public/librealsense-2.53.1/wrappers/open3d/cpp/RealSenseBagReader.cpp:162: undefined reference to `open3d::t::io::RSBagReader::NextFrame()'
/usr/bin/ld: /home/o/Public/librealsense-2.53.1/wrappers/open3d/cpp/RealSenseBagReader.cpp:158: undefined reference to `open3d::t::io::RSBagReader::IsEOF() const'
/usr/bin/ld: /home/o/Public/librealsense-2.53.1/wrappers/open3d/cpp/RealSenseBagReader.cpp:200: undefined reference to `open3d::t::io::RSBagReader::Close()'
/usr/bin/ld: /home/o/Public/librealsense-2.53.1/wrappers/open3d/cpp/RealSenseBagReader.cpp:201: undefined reference to `open3d::t::io::RSBagReader::~RSBagReader()'
/usr/bin/ld: /home/o/Public/librealsense-2.53.1/wrappers/open3d/cpp/RealSenseBagReader.cpp:201: undefined reference to `open3d::t::io::RSBagReader::~RSBagReader()'
collect2: error: ld returned 1 exit status
make[2]: *** [wrappers/open3d/CMakeFiles/RealSenseBagReader.dir/build.make:98: wrappers/open3d/RealSenseBagReader] Error 1
make[1]: *** [CMakeFiles/Makefile2:1334: wrappers/open3d/CMakeFiles/RealSenseBagReader.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

MartyG-RealSense, 2022-12-13:
Hi @viisautta  Are you following the guide at the link below and building RSBagReader using the built-in librealsense in Open3D instead of installing librealsense and its Open3D wrapper, please?

http://www.open3d.org/docs/latest/tutorial/sensor/realsense.html

viisautta, 2022-12-16:
@MartyG-RealSense Thanks. On your advice, building is ok as below:
...
[ 50%] Built target rs-kinfu
[ 50%] Built target RealSenseBagReader
[ 51%] Built target RealSenseRecorder
...

closed 2022-12-16 by viisautta

MartyG-RealSense, 2022-12-16:
It's excellent to hear that you were successful, @viisautta - thanks very much for the update!