realsenseai/librealsense #11210: Issues Building Librealsense from source on Kernel 5.10 Ubuntu 20.04

dverstandig, 2022-12-13:
Hello, 

I am trying to built librealsense from source on Jetpack 5.0 on a Jetson device. 

I am following the steps here: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

On step 4, I am running .`/scripts/patch-realsense-ubuntu-lts.sh`
 But I am met with this issue:

`Unsupported kernel version 5.10.104-tegra . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only`

Is there plans to support kernel 5.10 to be able to successfully built the librealsense on Jetpack 5.0 on the jetson? If not, what are your recommended paths forward to use the realsense on Jetpack 5.0?

Thanks!

MartyG-RealSense, 2022-12-13:
Hi @dverstandig  Kernel 5.10 is already supported, as Jetson devices have their own librealsense installation instruction page and a dedicated patch script called `./scripts/patch-realsense-ubuntu-L4T.sh`

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

dverstandig, 2022-12-13:
Thanks for the quick response. 

Using the script above, I now get this issue:

`Unsupported JetPack revision 35.1 aborting script`

This is running on a Jetson Xavier-NX on Jetpack 5.0

Thank you for the help!

MartyG-RealSense, 2022-12-13:
Although JetPack 5 is not officially supported at the time of writing this (it is currently being worked on for a future SDK release), RealSense users have been able to successfully use color and depth with Jetpack 5 and 5.02.  It has been reported that the IMU does not work with JetPack 5 on RealSense cameras equipped with an IMU though.

Advice for installing librealsense with JetPack 5 is provided by Jetson users at https://github.com/IntelRealSense/librealsense/issues/10416

dverstandig, 2022-12-15:
Thank you. 
I was able to get the realsense node working on noetic by building with `DFORCE_RSUSB_BACKEND=false`. I also had to downgrade the `libopencv-dev` version from 4.5 to 4.2 in order for the real sense node to not crash on launch. 

My steps were:
```
sudo git clone --depth 1 --branch v2.50.0 https://github.com/IntelRealSense/librealsense.git
cd librealsense
./scripts/setup_udev_rules.sh
sudo mkdir -p build && cd build
sudo cmake .. -DFORCE_RSUSB_BACKEND=false -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_CUDA=false
sudo make clean && sudo make && sudo make install
sudo apt-get install -q -y --no-install-recommends libopencv-dev=4.2.0+dfsg-5
```

Then, building the realsense-ros repo from source using the `ros1-legacy` branch at latest commit `44a703ed0ce7570668fae47e354a0b8d62ed1c05`

This is running on Jetpack 5.0, Kernel 5.10 ubuntu 20.04 running on noetic

MartyG-RealSense, 2022-12-15:
It's great to hear that you were successful.  Thanks so much for sharing your JetPack 5 / Ubuntu 20.04 installation solution with the RealSense community!

closed 2022-12-15 by dverstandig