realsenseai/librealsense #11211: Unity multiple human detection and tracking

pokemon9757, 2022-12-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                      | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                   | 
| Platform                                 | PC  |
| SDK Version                            |  2                        | 
| Language                            |  Unity C# | 
| Segment			|  Others                          | 

### Issue Description
Hello. I am trying to create an application that performs specific actions when **multiple users** get into the defined range of the camera. I need to track the distance from the camera to each identified individual and their position to perform different actions accordingly. 

I have spent the last entire 7 hours reading through all I could find on the documentation and past issues but still could not find anything specific for Unity in tracking the human body. I know there is a Unity wrapper sample but I haven't figured out how it works with Unity because of all the shaders thing. There really is a lack of examples in Unity that I could use. Please help me. Thank you very much!

MartyG-RealSense, 2022-12-13:
Hi @pokemon9757  If you are able to use a commercial solution in your project then the RealSense-compatible **Nuitrack SDK** tool that has Unity integration and supports multiple bodies may be a suitable foundation.

**Nuitrack SDK**
https://nuitrack.com/

**Nuitrack SDK body tracking example**
https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_RGB_Skeletons.md#displaying-the-skeleton

Nuitrack SDK is separate from the RealSense SDK software and does not interact with it.

***********

Alternatively you could explore the possibility of creating your own RealSense body tracking system for Unity, though this approach is more complex.  Intel have a YouTube seminar for doing this with RealSense at the link below. 

https://www.youtube.com/watch?v=VSHDyUXSNqY

[image]

pokemon9757, 2022-12-14:
Thanks a lot for your quick response! Unfortunately I am not expected to spend on any commercial product for this project... The alternative approach appears to involve a lot of neuron network training kind of work, which I have zero experience in either... 

Would you happen to know of any open-source library that could perform body tracking?

MartyG-RealSense, 2022-12-14:
An option for an open source body tracking solution with a Unity plugin is **OpenPose**.

https://github.com/CMU-Perceptual-Computing-Lab/openpose

https://www.intelrealsense.com/skeletal-tracking/

pokemon9757, 2022-12-14:
Thanks a lot! I will give those options a try and post a reply to describe which of them works later :)

MartyG-RealSense, 2022-12-14:
You are very welcome!  I recommend focusing on openpose, as the **Cubemos** product mentioned in the skeletal-tracking article is no longer available.

MartyG-RealSense, 2022-12-20:
Hi @pokemon9757  Do you require further assistance with this case, please?  Thanks!

pokemon9757, 2022-12-22:
Hello @MartyG-RealSense, thanks a lot for your recommendations! 

It turns out [Nuitrack SDK](https://nuitrack.com/) has a trial version and works perfectly well with Unity, they even have detailed demos! The remaining thing for me to do is to persuade my boss to get the premium license X'D

Merry Christmas and Happy new year!

closed 2022-12-22 by pokemon9757

MartyG-RealSense, 2022-12-22:
You are very welcome, @pokemon9757 - thanks very much for the well wishes, and I wish the same for you  :)