realsenseai/librealsense #11212: Depth stream pixels offset  pixel from Color stream

VishnuRamTK, 2022-12-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

VishnuRamTK, 2022-12-14:
Hi @MartyG-RealSense I am not sure how i can assign depth intrinsic the same values as color intrinsic. 
this is what i tried -
    
    rs2::align align_to_color(RS2_STREAM_COLOR);
    frames = p.wait_for_frames();
    frames = align_to_color.process(frames);
    rs2::frame color = frames.get_color_frame();
    rs2::frame depth = frames.get_depth_frame();
    auto depth_coloured = c.colorize(depth);
    auto color_profile = color.get_profile();
    auto color_stream_profile = color_profile.as<rs2::video_stream_profile>();
    auto color_intrin = color_stream_profile.get_intrinsics();

MartyG-RealSense, 2022-12-14:
You do not need to assign the depth intrinsic the same value as the color intrinsic.  When using the **align_to** instruction to align depth to color, like you are in your script above, the color intrinsic will be used **instead** of the depth intrinsic.  Your script looks correct to me.

VishnuRamTK, 2022-12-15:
In the mean time when i try to get the least distance from the whole depth stream i get an error -
rs::invalid_value_error what(): "out of range value for argument "y"

   for(int l = 0; l <= dep_width; l++)
     {	
      	for(int k = 0; k <= dep_height; k++ )
	{
		float dist = depth.as<rs2::depth_frame>().get_distance(l,k);
		temp = dist;
		if(temp < min_dist)
		{
			min_dist = temp;
		}
	}

MartyG-RealSense, 2022-12-15:
Does your script apply any post-processing filters?  There was a case at https://github.com/IntelRealSense/librealsense/issues/7395 that had this error when using filters.

VishnuRamTK, 2022-12-15:
Hi @MartyG-RealSense I am not using any filters. here is the full code on what i am trying to do -

#include <iostream>
#include <opencv2/core.hpp>
#include <opencv2/highgui.hpp>
#include <opencv2/opencv.hpp>
#include <opencv2/imgproc.hpp>
#include <vitis/ai/refinedet.hpp>
#include <librealsense2/rs.hpp>
#include <exception>
#include <string>
using namespace std;

int main( int argc, char *argv[] )
{
  
  // Create a Pipeline - this serves as a top-level API for streaming and processing frames
  float temp, min_dist = 10.0;
  rs2::pipeline p;
  rs2::config cfg;
  rs2::colorizer c;
  cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
  cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
  p.start(cfg);
  
  auto person_detect = vitis::ai::RefineDet::create("refinedet_pruned_0_96");
  rs2::align align_to_color(RS2_STREAM_COLOR);
  rs2::align align_to_depth(RS2_STREAM_DEPTH);
  rs2::frameset frames;
  
  for(int i = 0; i < 30; i++)
    {
        //Wait for all configured streams to produce a frame
        frames = p.wait_for_frames();
    }

  while (1)
  {
    
    frames = p.wait_for_frames();
    rs2::frame color = frames.get_color_frame();
    auto color_profile = color.get_profile();
    auto color_stream_profile = color_profile.as<rs2::video_stream_profile>();
    auto color_intrin = color_stream_profile.get_intrinsics();
    frames = align_to_color.process(frames);
    rs2::frame depth = frames.get_depth_frame();
    auto depth_coloured = c.colorize(depth);

    float dep_width = depth.as<rs2::video_frame>().get_width();
    float dep_height = depth.as<rs2::video_frame>().get_height();
    float col_width = color.as<rs2::video_frame>().get_width();
    float col_height = color.as<rs2::video_frame>().get_height();
    cv::Mat color_frame(cv::Size(col_width, col_height), CV_8UC3, (void*)color.get_data(),cv:: Mat::AUTO_STEP);
    cv::Mat depth_frame(cv::Size(dep_width, dep_height), CV_8UC3, (void*)depth_coloured.get_data(), cv::Mat::AUTO_STEP);

    auto detections = person_detect->run(color_frame);
    float width = col_width; //frame.cols;
    float height = col_height; //frame.rows;
    for (auto &roi : detections.bboxes)
    {
     int xmin = std::min(std::max(roi.x * width, 0.0f), width);
     int ymin = std::min(std::max(roi.y * height, 0.0f), height);
     int xmax = std::min(std::max(xmin + (roi.width * width), 0.0f), width);
     int ymax = std::min(std::max(ymin + (roi.height * height), 0.0f), height);

     cv::rectangle( color_frame,
                    cv::Point(xmin, ymin),
                    cv::Point(xmax, ymax),
                    cv::Scalar(0,255,0),
                    1,
                    1,
                    0 );
     cv::rectangle( depth_frame,
		     cv::Point(xmin, ymin),
		     cv::Point(xmax, ymax),
		     cv::Scalar(0,255,0),
		     1,
		     1,
		     0 );
    
     //get disance to the centre of detected object
     float dist = depth.as<rs2::depth_frame>().get_distance((xmax-xmin)/2,(ymax-ymin)/2);
     string s = to_string(dist);
     for(int l = 0; l <= dep_width; l++)
     {	
      	for(int k = 0; k <= dep_height; k++ )
	{
		float dist = depth.as<rs2::depth_frame>().get_distance(l,k);
		temp = dist;
		if(temp < min_dist)
		{
			min_dist = temp;
		}
	}
		
       
     }     
    
     cout << "distance is" << min_dist << "\n";
     cv::putText( color_frame,
		     s, 
		     cv::Point(xmin,ymin),
		     cv::FONT_HERSHEY_DUPLEX,
		     1.0,
		     CV_RGB(255, 0, 0),
		     2);
    }
    min_dist = 10.0;
    //cv::namedWindow("Person_Detection", cv::WINDOW_AUTOSIZE );
    cv::imshow("Person Detection", color_frame);
    cv::imshow("depth_Frame", depth_frame);
    if (cv::waitKey(1) == 'q') break;

  }

  return 0;
}

What i could observe is that the distance estimates of the detected object is accurate only when the object is present on the left end of the camera. if there is any error in the above code you can let me know. The object detection happens flawlessly with the color stream which i have already tested.

MartyG-RealSense, 2022-12-15:
What distance does **min_dist = 10.0;** represent, please?  Is it 10 cm or 10 meters?

VishnuRamTK, 2022-12-15:
It is in meters as the get_distance() function returns the depth of the pixel in meters. min_dist is just given a initial value to set some condition.

MartyG-RealSense, 2022-12-15:
What happens if instead of using this code for defining l and k:

```
for(int l = 0; l <= dep_width; l++)
{
for(int k = 0; k <= dep_height; k++ )
{
```

 you test with simpler code for l and k?

```
 int l = depth.get_width() / 2;
 int k = depth.get_height() / 2;
```

VishnuRamTK, 2022-12-17:
Hi @MartyG-RealSense everything works fine after matching the frames. The pixel point where i was measuring the depth was incorrect in the code.

closed 2022-12-17 by VishnuRamTK

MartyG-RealSense, 2022-12-17:
That's great, thanks very much for the update!