realsenseai/librealsense #11226: librealsense python wrapper build with cuda pipeline or postprocessing staying in cuda memory causing illegal memory access.

MartinPedersenpp, 2022-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Ubuntu 18 (L4T nvidia)  | 
| Kernel Version (Linux Only)    |  4.9                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX
| SDK Version                            |  2.51.1                     | 
| Language                            | python                  | 
| Segment			|  others                        | 

I am trying to move my working python script from my PC to my Jetson Xavier NX16 running jetpack 4.6 (cuda 10.2).
I have built my librealsense and python bindings with cuda and it works fine.
In my script, I am performing post processing on the depth data for it to be more suitable for analysis in Open3D later on.
As @MartyG-RealSense has already explained to me before, the python bindings are performing the post processing on the host CPU or GPU instead of on the on board processor of the camera. This is acceptable and I can see a clear performance difference where cuda is better than the ARM processor of the jetson.
I am, however, running into some issues when I want to utilize my GPU for other purposes after the RGB/depth frames have been captured.
I use Pytorch to do object detection on the RGB data and sometimes (no pattern) when Pytorch calls Torch.cuda.synchronize() before feeding data to the GPU, I get an Illegal memory access error, which causes everything to crash.

Is it possible that the Cuda processing of pyrealsense is happening Asynchronously or something, since it does not seem to leave the GPU after I move on from post processing?

I haven't done extensive testing of this, but it seems like adding a 0.05second delay after getting the frames and performing post processing resolves the issue, but ideally, the cuda processes should be able to manage this internally?

I will provide a example to reproduce if necessary, but I don't assume that you have a Xavier NX16 lying around to test it on.

MartyG-RealSense, 2022-12-16:
Hi @MartinPedersenpp  Googling for the term **pytorch cuda illegal memory access** shows in the search results that this error has occurred for a number of people in non-RealSense cases, so pyrealsense2 may not be causing the issue.

Some PyTorch issues with this issue found that updating PyTorch to its latest version resolved it.

MartinPedersenpp, 2022-12-16:
> Hi @MartinPedersenpp Googling for the term **pytorch cuda illegal memory access** shows in the search results that this error has occurred for a number of people in non-RealSense cases, so pyrealsense2 may not be causing the issue.
> 
> Some PyTorch issues with this issue found that updating PyTorch to its latest version resolved it. For other users, a combination of the latest PyTorch version and updating to a CUDA 11 version resolved it.

@MartyG-RealSense If I use librealsense built without cuda, the error is not present.
I noticed in my code that the alignment of my RGB/Depth data was happening in a loop which in turn caused some other error. Therefore I removed the alignment until I needed to capture the present frame of data. Without alignment, the post processing was happening way faster than with the alignment. Not sure if this was causing issues in the GPU.
I tried moving my assigning of pyrealsense.align() outside of my pipeline loop and only doing the actual alignment in the pipeline loop.
This seems to have resolved the issue, but I have no clue as to why.

MartyG-RealSense, 2022-12-16:
It's great to hear that you achieved a solution.

Align should be outside of a loop, otherwise it will keep re-instantiating inside the loop and slowing the application down.  Likewise, it is not recommended that pointclouds are created within a loop.

closed 2022-12-19 by MartinPedersenpp