realsenseai/librealsense #11227: Does Photo3D not recognize D455？

oscarWlde, 2022-12-16:
In win11，RealSense viewer can recognize D455，but Photo3D cannot.Is Photo3D unable to adapt to D455？

MartyG-RealSense, 2022-12-16:
Hi @oscarWlde  Is the link below the RealSense **Photo3D** application that you are using, please?

https://github.com/IntelRealSense/Photo3D

If so then as this project is over 3 years old it may not be currently designed for the D455 model (which was introduced in 2020).  The line from the project linked to below checks if the camera is a D415 or D435 model and activates a 'no camera plugged in' state if the model name does not match these.  If the model name does match then the program is permitted to continue running.

https://github.com/IntelRealSense/Photo3D/blob/master/Assets/Scripts/ContextDetector.cs#L48

Code that checks if the camera model name is D435 is also here:

https://github.com/IntelRealSense/Photo3D/blob/eed227fe404c9079f9e714683ffc6223e65d4ea9/Assets/Scripts/ContextManager.cs#L23-L34

A check for the terms 'D415' and 'D435' in the project indicates that these are the only sections of the project that check for a specific camera model, so conceivably you could try editing the scripts to refer to **D455** instead of D435.

oscarWlde, 2022-12-16:
Ok ,thanks.I know the reason.But I cannot develop by Unity,so it is hard for me to build up a new Unity App after changing  camera model name to D455. You'd better update your Photo3D application.

MartyG-RealSense, 2022-12-16:
I think that if I wanted to import this project into Unity, I would approach it like this:

1.  Left-click on the **Code** button on the project GitHub and select the **Download Zip** option to download the project as a compressed zip file.

[image]

2.  Extract the project folder **Photo3D-master** from the zip file.

3.  Open Unity and go to *File > Open Project** and select the unzipped folder to import the project into Unity.  It will likely spend some time updating the project's old files to be compatible with the Unity version that you are using.

4.  Double-leftclick on the two script files mentioned above and edit their D435 reference to D455.

5.  Run the project.

MartyG-RealSense, 2022-12-22:
Hi @oscarWlde  Do you require further assistance with this case, please?  Thanks!

oscarWlde, 2022-12-23:
> Hi @oscarWlde Do you require further assistance with this case, please? Thanks!
No，thanks for your help！

MartyG-RealSense, 2025-11-17:
Hi @oscarwilde666  I am not aware of the identity of the RealSense team member with admin powers, unfortunately.  Only an admin can delete an issue.  

The best that you may be able to do is left-click on the three dots at the end of each comment that you made and select Delete from the drop-down menu on each one until all your comments are removed.  I will do the same for my own comments if that is what you wish.

However, there are so many issues on this repository that it is extremely unlikely that someone would find your post, especially if you edit the title.