realsenseai/librealsense #11228: Setting new exposure only active in new frame after many frames

Siegel20, 2022-12-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    |                              | 
| Platform                                 | PC 
| SDK Version                            |  2.0                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
Using the Pythyon Intel RealSense SDK 2.0 API, I experience trouble with setting a new exposure on D435i (color camera) via 
sensor.set_option(rs.option.enable_auto_exposure, value)

I use only values such as 1, 2, 4, 9, 19, 39, 78, ..., 10000. 

- The streams depth and color are both active with 720 x 1280 @ 30 FPS. 
- Auto exposure is disabled for both streams.

When I receive a frameset via pipe.wait_for_frames(10000) and read the metadata of the color frame (frame_metadata_value.actual_exposure), it takes up to some 50 frames until the new value is active on the cam. I also checked the brightness of the received color frame and yes, the metadata fits to the brightness (I mean, if the meta data tells me a low exposure, the brightness is really low in the image). Often, it takes many frames (with increasing frame counters according to the meta data) until after some "random" time, the new exposure was applied. 

I want to implement a manual auto exposure ...

MartyG-RealSense, 2022-12-16:
Hi @Siegel20  When setting a manual exposure value for the color camera, may I confirm that you are addressing the RGB sensor instead of the depth sensor by using an index number of '1' with query_sensors() or using **first_color_sensor** to find the RGB sensor?  Otherwise, the **depth** auto-exposure value would be altered by default instead of RGB exposure.

**Index of '1'**

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
profile = pipeline.start(config)
color_sensor = profile.get_device().query_sensors()[1]
```

**first_color_sensor**

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
profile = pipeline.start(config)
color_sensor = profile.get_device().first_color_sensor()
```

***********

In regards to the time taken for the change in exposure to complete, the expectation would be that it would take a couple of frames for the firmware to register the exposure change and another 60 to 120 ms for the frames to get back to the algorithm.  This is confirmed at https://github.com/IntelRealSense/librealsense/issues/2104#issuecomment-411458598

In the brackets of wait_for_frames, a value as large as 10000 is almost never used.  What is the timing of your application like if you remove a value from the brackets and just have **wait_for_frames()** please?

Which RGB value are you using please?  Although the scale supports exposure to be set to up to 10000, in practice such large values are never used for RGB exposure.  It is more common to **reduce** it below its default, such as from 150 to 78.  I cannot recall having ever seen it set higher than 300.

MartyG-RealSense, 2022-12-22:
Hi @Siegel20  Do you require further assistance with this case, please?  Thanks!

Siegel20, 2022-12-22:
Hello @MartyG-RealSense 

thanks for your support. Let me get back to your questions raised. 
- Yes, I definitely worked with the exposure settings of the RGB camera. I checked this by reading stream name ("Color") from the profiles
- I removed the 10000 from wait_for_frames without any change in the timing behaviour

Before implementing a manual auto exposure using the RAW data (1920 x 1080 @ 30 FPS) of the color sensor, I wanted to assure that frame received via wait_for_frames() actually have the manually set exposure active (otherwise the algorithm would go crazy). 

Here are some results of my tests:

Test A
color sensor:  AE off, RAW data (1920 x 1080 @ 30 FPS) 
depth sensor: off
A1: 
- alternating 1 us and **156** us for the color sensor 
- up to **5** frames for new color exposure settings to be active, approx. **150** ms for receiving a frame with desired exposure
A2. 
- alternating 1 us and **625** us for the color sensor 
- up to **5** frames for new color exposure settings to be active, approx. **190** ms for receiving a frame with desired exposure
A3. 
- alternating 1 us and **1250** us for the color sensor 
- up to **6** frames for new color exposure settings to be active, approx. **450** ms for receiving a frame with desired exposure

Test B
color sensor:  AE off, RAW data (1920 x 1080 @ 30 FPS) 
depth sensor: AE on, Z16 data (1280 x 720 @ 30 FPS)
B1: 
- alternating 1 us and **156** us for the color sensor 
- up to **3** frames for new color exposure settings to be active, approx. **390** ms for receiving a frame with desired exposure
B2. 
- alternating 1 us and **625** us for the color sensor 
- up to **6** frames for new color exposure settings to be active, approx. **510** ms for receiving a frame with desired exposure
B3. 
- alternating 1 us and **1250** us for the color sensor 
- up to **10** frames for new color exposure settings to be active, approx. **580** ms for receiving a frame with desired exposure

Case A1 is okay, but of course it would be great if I would get down with these times (esp. when 3D is on as well => B1). 

Is it true that manual exposure allows only dedicated exposure times (1, 2, 4, 9, 19, 39, 78, 156, 312, 625, 1250, 2500, 5000, and 10000) whereas the internal auto exposure can set other times (at least it seems so)? Would it make sense to use the gain for a fine adjustment in manual exposure mode?

Thanks for your advice!

MartyG-RealSense, 2022-12-22:
Yes, RGB manual exposure only accepts the values that you listed.  If a value other than those is entered then the SDK uses the value that is closest to an accepted value.  For example, using '12' automatically sets '9' as the value, whilst using '17' causes the value to be set as '19'.  

You can test this principle by inputting values into the RGB exposure setting in the RealSense Viewer tool.  As demonstrated in the images below, manually inputting '290' by left-clicking on the pencil icon next to the option to type in a value on the keyboard causes '312' to be selected as the exposure value.

[image]

[image]

A RealSense team member explains at https://github.com/IntelRealSense/librealsense/issues/5493#issuecomment-567857540 that the extent to which exposure values can be fine-controlled depends on whether your computer is using Windows or Linux.  The Media Foundation backend on Windows will round up values.

There is the option to build the RealSense SDK from source code on Windows with CMake instead of installing it with the **Intel.RealSense.SDK-WIN10** installer file.  If the CMake build flag **FORCE_RSUSB_BACKEND** is set to True then the SDK should use a WinUSB rewrite of the UVC protocol as its backend instead of Media Foundation.

Siegel20, 2023-01-02:
Hello @MartyG-RealSense,

thanks for you assistance. I will close the issue now and I have to live with the fact that the cam has some unwanted effects.

closed 2023-01-02 by Siegel20

MartyG-RealSense, 2023-01-02:
You are very welcome, @Siegel20 - thanks very much for your patience and for the update.