realsenseai/librealsense #11229: How save ply by pyrealsense2 with color Texture rather than depth Texture?

chibai, 2022-12-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I want to save the point cloud as follow, which is textured by color as belowe
[image]
But according the example https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py
I can only get point cloud as follow, I tried many approaches, but failed
[image]

MartyG-RealSense, 2022-12-17:
Hi @chibai  The only Python export_to_ply script that is known to be able to successfully export color is at https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293

Any investigation about saving a color ply with Python always ends up coming back to the export_to_ply example linked to above unfortunately, as it is the only confirmed example that works.  

There is also a **save_to_ply** example script at https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152 which exports depth but did not export color for the script's creator even though it was intended to.  As it has been a couple of years since the script was last tried, you are welcome to try it yourself to see whether you get different results now.

chibai, 2022-12-17:
> Hi @chibai The only Python export_to_ply script that is known to be able to successfully export color is at [#6194 (comment)](https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293)
> 
> Any investigation about saving a color ply with Python always ends up coming back to the export_to_ply example linked to above unfortunately, as it is the only confirmed example that works.
> 
> There is also a **save_to_ply** example script at [#7747 (comment)](https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152) which exports depth but did not export color for the script's creator even though it was intended to. As it has been a couple of years since the script was last reported to have been tried, you are welcome to try it yourself to see whether you get different results now.

Thank you for your fast reply~~I will try to dig a little bit, before any new results, I will close this issue~

closed 2022-12-17 by chibai