realsenseai/librealsense #11230: Invalid Value in rs2_get_option

DukeMacrocat, 2022-12-19:
| | |
|---|---|
|**librealsense**|2.51.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D455|
|**Serial Number**|215122256443|
|**Firmware Version**|05.14.00.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|211523064561|
|**Firmware Update Id**|211523064561|

Please provide a description of the problem

Invalid Value in rs2_get_option(options:0x7f239c0011f0, option:Auto Gain Limit
Toggle):
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)

Please help to solve the issue.

MartyG-RealSense, 2022-12-19:
Hi @DukeMacrocat  There was a recently reported case of this error at the link below when using firmware **5.14.0.0**.  Downgrading to the previous firmware version **5.13.0.50** resolved the issue.

https://support.intelrealsense.com/hc/en-us/community/posts/12079922677907

Nir-Az, 2022-12-20:
Hi @DukeMacrocat ,
This option was deprecated on FW 5.14.0
You will have to work with the new release of the SDK for nor seeing this issue, the change is a bundle change
Please try SDK 2.54.0 and update.

DukeMacrocat, 2022-12-27:
@MartyG-RealSense @Nir-Az Thank you i solved the problem

MartyG-RealSense, 2022-12-27:
That's excellent news that you achieved a solution, @DukeMacrocat - thanks very much for the update!

MartyG-RealSense, 2022-12-31:
Case closed due to solution achieved.

closed 2022-12-31 by MartyG-RealSense

AndreV84, 2023-01-25:
where do we get sdk 2.54.0? it hasn't been released
@MartyG-RealSense

MartyG-RealSense, 2023-01-25:
The **development** branch of the SDK that becomes the next 'latest' SDK when released can be found at the link below.

https://github.com/IntelRealSense/librealsense/tree/development

czp0204, 2023-05-01:
Hi, I'm using a D435i with ubuntu LTS on Jetson nano. Sorry to bother, could you tell me how to downgrade the firmware from 5.14 to 5.13 ?

LiZhi-sus, 2023-06-28:
> @MartyG-RealSense @Nir-Az Thank you i solved the problem

I met the same problem. How do you solve this problem?

LiZhi-sus, 2023-06-28:
I met the same problem. How do you solve it?