realsenseai/librealsense #11236: installing librealsense on Ubuntu 22.04

monajalal, 2022-12-19:
Hi,

I have a D435 camera and want to install librealsense and got the following error. Could you please guide me how to fix this?

```
[mona@ARD-WS-24 20221219-15:57:32]$ ./bootstrap-vcpkg.sh
Downloading vcpkg-glibc...
Telemetry
---------
vcpkg collects usage data in order to help us improve your experience.
The data collected by Microsoft is anonymous.
You can opt-out of telemetry by re-running the bootstrap-vcpkg script with -disableMetrics,
passing --disable-metrics to vcpkg on the command line,
or by setting the VCPKG_DISABLE_METRICS environment variable.

Read more about vcpkg telemetry at docs/about/privacy.md
[mona@ARD-WS-24 20221219-15:57:38]$ ./vcpkg integrate install
Applied user-wide integration for this vcpkg root.
CMake projects should use: "-DCMAKE_TOOLCHAIN_FILE=/home/mona/vcpkg/scripts/buildsystems/vcpkg.cmake"
[mona@ARD-WS-24 20221219-15:57:51]$ ./vcpkg install realsense2
Computing installation plan...
A suitable version of cmake was not found (required v3.25.0) Downloading portable cmake 3.25.0...
Downloading cmake...
https://github.com/Kitware/CMake/releases/download/v3.25.0/cmake-3.25.0-linux-x86_64.tar.gz->/home/mona/vcpkg/downloads/cmake-3.25.0-linux-x86_64.tar.gz
Extracting cmake...
The following packages will be built and installed:
  * libusb[core]:x64-linux -> 1.0.26#2
    realsense2[core]:x64-linux -> 2.51.1
  * vcpkg-cmake[core]:x64-linux -> 2022-10-30
  * vcpkg-cmake-config[core]:x64-linux -> 2022-02-06#1
Additional packages (*) will be modified to complete this operation.
Detecting compiler hash for triplet x64-linux...
Restored 0 package(s) from /home/mona/.cache/vcpkg/archives in 10.29 us. Use --debug to see more details.
Installing 1/4 libusb:x64-linux...
Building libusb[core]:x64-linux...
libusb currently requires the following tools and libraries from the system package manager:
    autoreconf
    libudev

These can be installed on Ubuntu systems via apt-get install autoreconf libudev-dev
-- Downloading https://github.com/libusb/libusb/archive/4239bc3a50014b8e6a5a2a59df1fff3b7469543b.tar.gz -> libusb-libusb-4239bc3a50014b8e6a5a2a59df1fff3b7469543b.tar.gz...
-- Extracting source /home/mona/vcpkg/downloads/libusb-libusb-4239bc3a50014b8e6a5a2a59df1fff3b7469543b.tar.gz
-- Using source at /home/mona/vcpkg/buildtrees/libusb/src/3b7469543b-a464ae495e.clean
-- Getting CMake variables for x64-linux-dbg
-- Getting CMake variables for x64-linux-rel
CMake Error at scripts/cmake/vcpkg_configure_make.cmake:597 (message):
  libusb requires autoconf from the system package manager (example: "sudo
  apt-get install autoconf")
Call Stack (most recent call first):
  ports/libusb/portfile.cmake:61 (vcpkg_configure_make)
  scripts/ports.cmake:147 (include)

error: building libusb:x64-linux failed with: BUILD_FAILED
error: Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
    https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+libusb
You can submit a new issue at:
    https://github.com/microsoft/vcpkg/issues/new?template=report-package-build-failure.md&title=[libusb]+Build+error
Include '[libusb] Build error' in your bug report title, the following version information in your bug description, and attach any relevant failure logs from above.
    vcpkg-tool version: 2022-12-14-7ae0d8527fb488fde10a89c2813802dc9b03b6f9
    vcpkg-scripts version: a22a71181 2022-12-19 (40 minutes ago)

![Screenshot from 2022-12-19 16-02-09](https://user-images.githubusercontent.com/1892917/208523359-0d181c90-d3b2-44cd-bb55-bd633910ade0.png)

Please use the prefilled template from /home/mona/vcpkg/installed/vcpkg/issue_body.md when reporting your issue.

```

It suggests to use `vcpkg update` and when I use it, I get this error:

```
[mona@ARD-WS-24 20221219-16:03:29]$ ./vcpkg update
Using local portfile versions. To update the local portfiles, use `git pull`.
No packages need updating.
```

When I do `git pull` nothing happens since I just got the repo.

```
[mona@ARD-WS-24 20221219-16:03:34]$ git pull
Already up to date.
```

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

monajalal, 2022-12-19:
[mona@ARD-WS-24 20221219-16:11:35]$ sudo apt -y install dh-autoreconf
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  autoconf automake autopoint autotools-dev libdebhelper-perl libltdl-dev libtool m4
Suggested packages:
  autoconf-archive gnu-standards autoconf-doc gettext libtool-doc gfortran | fortran95-compiler
  gcj-jdk m4-doc
The following NEW packages will be installed:
  autoconf automake autopoint autotools-dev dh-autoreconf libdebhelper-perl libltdl-dev libtool m4
0 upgraded, 9 newly installed, 0 to remove and 52 not upgraded.
Need to get 1,977 kB of archives.
After this operation, 7,325 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 m4 amd64 1.4.18-5ubuntu2 [199 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 autoconf all 2.71-2 [338 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 autotools-dev all 20220109.1 [44.9 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 automake all 1:1.16.5-1.3 [558 kB]
Get:5 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 autopoint all 0.21-4ubuntu4 [422 kB]
Get:6 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libdebhelper-perl all 13.6ubuntu1 [67.2 kB]
Get:7 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libtool all 2.4.6-15build2 [164 kB]
Get:8 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 dh-autoreconf all 20 [16.1 kB]
Get:9 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libltdl-dev amd64 2.4.6-15build2 [169 kB]
Fetched 1,977 kB in 0s (4,582 kB/s)      
Selecting previously unselected package m4.
(Reading database ... 280606 files and directories currently installed.)
Preparing to unpack .../0-m4_1.4.18-5ubuntu2_amd64.deb ...
Unpacking m4 (1.4.18-5ubuntu2) ...
Selecting previously unselected package autoconf.
Preparing to unpack .../1-autoconf_2.71-2_all.deb ...
Unpacking autoconf (2.71-2) ...
Selecting previously unselected package autotools-dev.
Preparing to unpack .../2-autotools-dev_20220109.1_all.deb ...
Unpacking autotools-dev (20220109.1) ...
Selecting previously unselected package automake.
Preparing to unpack .../3-automake_1%3a1.16.5-1.3_all.deb ...
Unpacking automake (1:1.16.5-1.3) ...
Selecting previously unselected package autopoint.
Preparing to unpack .../4-autopoint_0.21-4ubuntu4_all.deb ...
Unpacking autopoint (0.21-4ubuntu4) ...
Selecting previously unselected package libdebhelper-perl.
Preparing to unpack .../5-libdebhelper-perl_13.6ubuntu1_all.deb ...
Unpacking libdebhelper-perl (13.6ubuntu1) ...
Selecting previously unselected package libtool.
Preparing to unpack .../6-libtool_2.4.6-15build2_all.deb ...
Unpacking libtool (2.4.6-15build2) ...
Selecting previously unselected package dh-autoreconf.
Preparing to unpack .../7-dh-autoreconf_20_all.deb ...
Unpacking dh-autoreconf (20) ...
Selecting previously unselected package libltdl-dev:amd64.
Preparing to unpack .../8-libltdl-dev_2.4.6-15build2_amd64.deb ...
Unpacking libltdl-dev:amd64 (2.4.6-15build2) ...
Setting up libdebhelper-perl (13.6ubuntu1) ...
Setting up m4 (1.4.18-5ubuntu2) ...
Setting up autotools-dev (20220109.1) ...
Setting up autopoint (0.21-4ubuntu4) ...
Setting up autoconf (2.71-2) ...
Setting up automake (1:1.16.5-1.3) ...
update-alternatives: using /usr/bin/automake-1.16 to provide /usr/bin/automake (automake) in auto mo
de
Setting up libtool (2.4.6-15build2) ...
Setting up dh-autoreconf (20) ...
Setting up libltdl-dev:amd64 (2.4.6-15build2) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for install-info (6.8-4build1) ...

closed 2022-12-19 by monajalal

monajalal, 2022-12-19:
^
sudo apt-get install libudev-dev

wangmingxintsinghua, 2025-03-18:
may i ask u whether you have solved this error？i meet the same error with you！ thank you！

efagerberg, 2025-03-31:
For me (6.11 kernel ubuntu) I was able to get passed this error with `sudo apt -y install dh-autoreconf`

MartyG-RealSense, 2025-04-01:
Thanks so much @efagerberg for sharing the method that worked for you in your own situation!