realsenseai/librealsense #11239: Using something similar to pyrealsense2.pipeline and wait_for_frames but with gstreamer or other for an rgb camera

MarioCavero, 2022-12-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | ? dmidecode 3.2 ? | 
| Operating System & Version |   Mendel Linux  | 
| Kernel Version (Linux Only)    |  4.14.98-imx  | 
| Platform                                 | Google Coral dev board  |
| SDK Version                            |  {}                      | 
| Language                            |  python   | 
| Segment			|          {}             | 

### Issue Description
I have been using the `pyrealsense2` library in python to obtain images and using python Pipe's to get messages to start and stop, and `realsense.pipeline` and `rs.wait_for_frames() `to capture and send images to a subscriber. It works fine. 

I have been trying to do it with several devices at the same time, but some won't have a RealSense camera, but a normal RGB camera (f.g coral camera). I haven't managed to make a pipeline system the same as realsense's one. I've read that [gstreamer](https://github.com/google-coral/examples-camera) can work fine, but I haven't been able to make it work. I guess it's due to how rs.pipeline works, differently to just open and get images with regular opencv functions like  
```cap = cv2.VideoCapture(args.camera_idx)

    while cap.isOpened():
        ret, frame = cap.read()
        if not ret:
            break
```

MartyG-RealSense, 2022-12-20:
Hi @MarioCavero  The RealSense SDK has the ability to access non-RealSense RGB cameras as a **Platform Camera**.  You can test this in the RealSense Viewer by left-clicking on the **Add Source** button at the top of the Viewer's options side-panel and selecting the **Platform Camera** sub-option.

[image]

[image]

In Python, you could differentiate between a RealSense camera and a non-RealSense camera by checking its **camera name**.  This is demonstrated in the multiple-camera supporting **realsense_device_manager.py** script at the link below that is part of the RealSense SDK's **box_dimensioner_multicam** project.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L45

The link below has another Python example of checking a camera's name.

https://snyk.io/advisor/python/pyrealsense2/functions/pyrealsense2.camera_info

[image]

*************

If you would prefer to use GStreamer, there are a couple of examples of RealSense plugins for GStreamer here: 

https://github.com/WKDSMRT/realsense-gstreamer

https://www.aivero.com/2020/03/gstreamer-elements-realsense-open-sourced/
https://gitlab.com/aivero/public/gstreamer/gst-realsense/-/releases

MarioCavero, 2022-12-20:
Nice, thanks for such a quick answer @MartyG-RealSense . I will try adding the platform camera, as just adding the code to detect a non-rgb camera does not show anything! I guess it needs to be "added and identified" first. The SDK seems to not work on embedded devices that do not have a screen (same as opencv cannot show images in devices that do not have a screen). I will work on this with the info you provided to try and get a work around! Again, thanks for a detailed and quick response!

MartyG-RealSense, 2022-12-20:
You are very welcome!

If you build the SDK from source code with CMake then you can configure it for 'headless' use without a display by including the build flag **-DBUILD_GRAPHICAL_EXAMPLES=false** in the CMake build instruction.  The build will then exclude OpenGL graphics support and graphical tools and examples.

MartyG-RealSense, 2022-12-30:
Hi @MarioCavero  Do you require further assistance with this case, please?  Thanks!

MarioCavero, 2023-01-01:
> 

Hi @MartyG-RealSense . Thanks for keeping up. I am working with this parallel to other stuff, so at the moment I can't update you with anything! I will re-open if I find any issues concerning this! Thank you very much for your quick responses and follow ups!

closed 2023-01-01 by MarioCavero

MartyG-RealSense, 2023-01-01:
You are very welcome, @MarioCavero - please do feel free to re-open this case at a future date if you need further assistance with it.  Good luck!