realsenseai/librealsense #11241: Export pointcloud info without mesh and with RGB info and only of a certain region

Ankush1909IIT, 2022-12-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Ankush1909IIT, 2022-12-20:
I am trying to export point cloud information of a region without mesh information. The image below is what I am looking to export. Point cloud with RGB and without mesh information.

And I am looking to be able to specify the domain (x,y,z distance from the camera and then a square bounding box domain) and be able to export the point cloud domain with RGB and without mesh.

Any help will be appreciated.

[image]

MartyG-RealSense, 2022-12-20:
Hi @Ankush1909IIT  Let's begin by tackling the pointcloud export part of your question first.  The Python and C++ script references for .ply export at https://github.com/IntelRealSense/librealsense/issues/6411#issuecomment-1024245007 may be helpful to you.

The Python script uses the **export_to_ply** instruction to export a color .ply file without vertex normals.

The C++ script uses the **save_to_ply** instruction and supports .ply export configuration options similar to those of the RealSense Viewer's .ply  export interface that you quoted in the image above.

Ankush1909IIT, 2022-12-20:
Hi @MartyG-RealSense , I am using python.

MartyG-RealSense, 2022-12-20:
If you would prefer to use save_to_ply to achieve configuration options similar to the Viewer, there is a Python save_to_ply example script at [#7747 (comment)](https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152) which exports depth but did not export color for the script's creator even though it was intended to.  The export_to_ply script that exports with color but without vertex normals is the only confirmed successful way to export a color .ply in Python.

As it has been a couple of years since the Python save_to_ply script was last reported to have been tried though, you are welcome to try it yourself to see whether you get different results now.

Ankush1909IIT, 2022-12-20:
Hi @MartyG-RealSense can you share the python code for export to ply option. I tried changing it in orignal code for export given as example and it return the error module 'pyrealsense2' has no attribute 'export_to_ply'.

MartyG-RealSense, 2022-12-20:
The RealSense SDK has a complete example Python program called **export_ply_example.py** for saving a .ply with **save_to_ply** though without color included, so you would have to use it as a reference to work out how to add color export with the help of the code in https://github.com/IntelRealSense/librealsense/issues/7747#issuecomment-725346152

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

In regard to export_to_ply, the SDK example **pyglet_pointcloud_viewer.py** uses this instruction for .ply export, though the overall program may be more complex than you need.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/pyglet_pointcloud_viewer.py#L491

Ankush1909IIT, 2022-12-20:
## License: Apache 2.0. See LICENSE file in root directory.
## Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#####################################################
##                  Export to PLY                  ##
#####################################################

# First import the library
import pyrealsense2 as rs

# Declare pointcloud object, for calculating pointclouds and texture mappings
pc = rs.pointcloud()
# We want the points object to be persistent so we can display the last cloud when a frame drops
points = rs.points()

# Declare RealSense pipeline, encapsulating the actual device and sensors
pipe = rs.pipeline()
config = rs.config()
# Enable depth stream
config.enable_stream(rs.stream.depth)

# Start streaming with chosen configuration
pipe.start(config)

# We'll use the colorizer to generate texture for our PLY
# (alternatively, texture can be obtained from color or infrared stream)
colorizer = rs.colorizer()

try:
    # Wait for the next set of frames from the camera
    frames = pipe.wait_for_frames()
    colorized = colorizer.process(frames)

    # Create save_to_ply object
    ply = rs.save_to_ply("1.ply")
   

    # Set options to the desired values
    # In this example we'll generate a textual PLY with normals (mesh is already created by default)
    ply.set_option(rs.save_to_ply.option_ply_binary, False)
    #ply.set_option(rs.save_to_ply.option_ply_normals, False)
    ply.set_option(rs.save_to_ply.option_ply_mesh, False)
    ply.set_option(rs.save_to_ply.option_ignore_color, False)
    ply.set_option(rs.save_to_ply.option_ply_threshold, False)
    #ply.set_option(rs.save_to_ply.option_ply_normals, True)

    print("Saving to 1.ply...")
    # Apply the processing block to the frameset which contains the depth frame and the texture
    ply.process(colorized)
    print("Done")
finally:
    pipe.stop()

Using this code now. It has stopped exporting mesh info but is not giving rgb colour information.

MartyG-RealSense, 2022-12-20:
The subject of color ply export with Python has been extensively researched several times with multiple references examined, and each time the conclusion has been that the only working method to export a color ply from Python is with the export_to_ply script at https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293 unfortunately.

Ankush1909IIT, 2022-12-20:
import numpy as np                     
import pyrealsense2 as rs             
import numpy as np

print("Environment Ready")

pc = rs.pointcloud()
pipe = rs.pipeline()

#Create a config and configure the pipeline to stream

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)

#Start streaming
profile = pipe.start(config)
align_to = rs.stream.color
align = rs.align(align_to)

for x in range(10):
	pipe.wait_for_frames()

frames = []
for x in range(2):
	frames = pipe.wait_for_frames()
	aligned_frames = align.process(frames)

	#Get aligned frames
	aligned_depth_frame = aligned_frames.get_depth_frame() 
	color_frame = aligned_frames.get_color_frame()
	
	points = pc.calculate(aligned_depth_frame)
	pc.map_to(color_frame)
	print("Saving to 1.ply...")
	points.export_to_ply("1.ply", color_frame)
print("Done")	

This code is working

Ankush1909IIT, 2022-12-20:
Do we have options to only export a region?

MartyG-RealSense, 2022-12-21:
You could implement a post-processing **threshold filter** in Python to exclude Z-depth data outside of a certain range.  References for doing so in Python can be found at https://github.com/IntelRealSense/librealsense/issues/5964#issuecomment-594372943

XY can have detail removed using a **bounding box** to remove all XY detail outside of the bounding box, though it is a complicated procedure.

I will be happy to continue this discussion with you tomorrow.  I will be back in 7 hours from now.  Good luck!

Ankush1909IIT, 2022-12-21:
yes, we can continue tomorrow.

MartyG-RealSense, 2022-12-28:
Hi @Ankush1909IIT  Do you require further assistance with this case, please?  Thanks!

Ankush1909IIT, 2022-12-28:
Hi @MartyG-RealSense , can you please keep it open? I am still working on it by looking at other question but yet to reach a solution.

MartyG-RealSense, 2022-12-28:
Okay, it's no problem to keep it open.  Thanks very much for the update!

Ankush1909IIT, 2023-01-10:
Hi @MartyG-RealSense , can you let me know if we have a solution to exporting without mesh. The code that I shared is exporting with color but as a mesh and not as point cloud? I looked at some of the answers but without any success yet.

MartyG-RealSense, 2023-01-11:
If using the Python script at https://github.com/IntelRealSense/librealsense/issues/6411#issuecomment-1024245007 to export a ply with color but without vertex normals is not suitable for you then there is not an alternative solution for Python that is known to work, unfortunately.

MartyG-RealSense, 2023-01-17:
Hi @Ankush1909IIT  Bearing in mind the above comment, do you require further assistance with this case please?  Thanks!

Ankush1909IIT, 2023-01-19:
import numpy as np                     
import pyrealsense2 as rs             
import numpy as np

print("Environment Ready")

pc = rs.pointcloud()
pipe = rs.pipeline()
#point = rs.points()

#Create a config and configure the pipeline to stream

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)

# Start streaming
profile = pipe.start(config)
align_to = rs.stream.color
align = rs.align(align_to)

threshold_filter = rs.threshold_filter()
threshold_filter.set_option(rs.option.max_distance, 0.4)
threshold_filter.set_option(rs.option.min_distance, 0.0)

for x in range(10):
	pipe.wait_for_frames()

frames = []
for x in range(2):
	frames = pipe.wait_for_frames()
	#ply = rs.save_to_ply("1.ply")
	frames_filtered = threshold_filter.process(frames)
	aligned_frames = align.process(frames)

	# Get aligned frames
	aligned_depth_frame = aligned_frames.get_depth_frame() 
	color_frame = aligned_frames.get_color_frame()
	
	points = pc.calculate(aligned_depth_frame)
	pc.map_to(color_frame)
	print("Saving to 1.ply...")
	points.export_to_ply("1.ply", color_frame)

print("Done")	

Hi, @MartyG-RealSense , I am using this code to remove the background. I am not sure why it is not working? Do you have any idea?

MartyG-RealSense, 2023-01-20:
I carefully checked your code and cannot find any obvious problems with your approach, and you are using the code at https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293 which is the only known working method of exporting color to ply.  So it should work.

What happens if you comment out these two lines:

```
frames = []
for x in range(2):
```

MartyG-RealSense, 2023-01-26:
Hi @Ankush1909IIT  Do you have an update about this case that you can provide, please?  Thanks!

Ankush1909IIT, 2023-01-31:
Hi @MartyG-RealSense - thanks for your help, the problem was not solved. However, I wrote my own code which reads the info line by line and removes the background. Thank you for your help. I will close this.

closed 2023-01-31 by Ankush1909IIT

MartyG-RealSense, 2023-02-01:
Thanks very much @Ankush1909IIT for the update!  I'm pleased to hear that you achieved a solution.