realsenseai/librealsense #11243: D415 "IC2 Config error" when attempting sync

teunhoevenaars, 2022-12-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.13.0.50 as suggested on https://dev.intelrealsense.com/docs/recommended-production-camera-configurations | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0 as suggested on https://dev.intelrealsense.com/docs/recommended-production-camera-configurations  | 
| Language                            |                           | 
| Segment			|  Robot                         | 

### Issue Description

In our application we have connected two RealSense D415 assemblies (i.e. Depth Module D415 + Vision Processor D4 Board) as Master and Slave. Additionally, in parallel to the connection between Master and Slave there is also a connection from Master to the input of a custom PCB, which is used to control our LEDs. While sometimes we manage to achieve a good sync between the two cameras (and the LEDs), in other occassions the sync is not achieved. Our application will run on Linux, but currently we're trying to get consistent behavior when using the **Intel.RealSense.Viewer**.

When using the Viewer, we regularly encounter errors of type "Left IC2 Config error" when starting capture on the Slave (the Master is already capturing at this point). Same behavior when switching Master and Slave roles between the cameras. We have difficulty finding the cause for this error. 

Questions we have are:
- what triggers this "Left IC2 Config error"? We find the error message vague. We are further confused by https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-private.h where this error is mapped to variable _i2c_cfg_left_error_ which seems to indicate that it is an i2c communication error
- could this error be the result of damage to the Vision Processor D4 Board? We have had one instance were this issue was resolved by replacing that board. If so, can the Vision Processor D4 Board be damaged by a wrong configuration, e.g. both RealSense modules configured as Master or unintended effects of our custom PCB connected in parallel?
- what is the lower limit of the trigger signal required by the Slave and could a a trigger signal that is too close to this lower limit  result in the indicated error? I'm asking because in an initial test architecture the voltage dropped below 1V8 and sync was not achieved, but not sure if this is related. Perhaps we're still too close to the lower limit although Fig. 17 on https://www.ti.com/lit/an/scya049a/scya049a.pdf?ts=1671573719357 seems to indicate that there is sufficient margin in 1V8 CMOS logic levels.
- Related to the point above; what is the current of the outgoing trigger signal from the Master? This allows is to set constraints on the load on our parallel input circuit.
- In one system we get immediate "Left IC2 Config error" when we're activating the Slave camera (in Mode 2), both using the **Intel.RealSense.Viewer** as well as using the C++ API on Linux Ubuntu. However, when we configure the Slave in Mode 3 (we know, unsupported) via the C++ API, then the system works fine. Can you tell us what is different in Mode 3 vs Mode 2 that would explain this?

Many thanks in advance!

teunhoevenaars, 2022-12-21:
Thanks @MartyG-RealSense for the quick response and for reposting the External Synchronization white-paper! However, I believe some of my questions have been left unanswered so please allow me to update my questions and their status:

1. _what triggers this "Left IC2 Config error"?_ This is likely an issue with the Vision Processor D4 Board as per https://github.com/IntelRealSense/realsense-ros/issues/895#issuecomment-527313596. 

- > We use a V1 version of the Vision Processor D4 Board
- > We enable master first and Slave second.
- > We make use of the trigger generated by the Master RealSense.
- > We are only connecting pins 5 and 9 as explained on https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration with 3 parallel connections: 1) a Master RealSense, 2) a Slave RealSense, and 3) a connection to an input of our Custom LED driver PCB. The Master RealSense is the only one generating a signal.
- > No clarification received regarding the meaning of "Left IC2 Config error" and whether this may be a typo (IC2 instead of I2C) as suggested by the mapping to _i2c_cfg_left_error_ as introduced on https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-private.h

2. _Could this error be the result of damage to the Vision Processor D4 Board?_ My impression is that it unlikely a hardware defect, because:
- > it only manifests itself in Slave mode and switches to another RealSense assembly when swapping Master and Slave roles
- >  the issue disappears when using Slave Mode 3 instead of Slave Mode 2. NB we do **not** intend to use Slave Mode 3 but we tried different settings to pinpoint the issue. The fact that switching to Slave Mode 3 (instead of 2) makes the issue disappear is surprising and so far unexplained.

3. _could the error be explained by a small voltage drop in the trigger signal?_ no direct answer provided, but assuming that RealSense input trigger behavior follows the levels as described on https://www.ti.com/lit/an/scya049a/scya049a.pdf?ts=1671573719357, then this is unlikely. We have not observed voltage drops larger than 0.1V, and therefore this explanation is not considered valid and for now discarded.

Looking forward to hearing back from you and thank you for your efforts.

MartyG-RealSense, 2022-12-21:
1.
Left IC2 Config error is the correct term.  It may be related to the interposer connection between the Vision Processor D4 Board and the depth module. 

What method are you using to connect the D4 board and the depth module board together - a flexible interposer cable or a rigid interposer PCB board?  

Can you check please that the interposer is firmly inserted into the 50-pin connection port at each end (the D4 end and the depth module end) and is not partially inserted?  If the connectors are fully inserted then this would help to rule out an intermittent problem arising from a partial insertion of the connections.

For example, the occasion where replacing the D4 resolved the problem may have been because the 50-pin connections were fully inserted on that occasion.

2.
As stated above, I would expect it to be an interposer issue rather than damage to the D4 board, and the camera is apparently able to function normally when Inter Cam Sync Mode is set to its default non-synced '0' mode, yes?

I have not previously encountered a case where a camera would have a problem when set to '2' but not when set to '3'.  I do not have an explanation for that.  

Does the error still occur if you only list one camera in the Viewer instead of two and start the depth stream with its Inter Cam Sync Mode set to '2', with no master camera connected to it?

3.  If the voltage of the trigger signal dropped then my expectation would be that the result would simply be that timestamp sync would not successfully occur, and it would not cause a problem with the camera hardware itself.

teunhoevenaars, 2022-12-22:
Thanks MartyG for clarifying the IC2/I2C confusion!
1. We're currently using a rigid interposer but considering to switch to a flexible interposer. We'll double-check if everything is connected properly in the next days and I'll get back to you. 
2. Cameras indeed function properly in '0' Mode. As far as I remember the cameras also function properly if operatied in '2' Mode with no master connected, but we'll double-check this too and I will update this response afterwards.
3. Thanks for sharing your insights!

MartyG-RealSense, 2022-12-22:
You are very welcome, @teunhoevenaars - thanks very much @teunhoevenaars for the update.  I look forward to your next report.  :)

teunhoevenaars, 2023-01-02:
Hi @MartyG-RealSense, 
I've investigated the interposer, but that is fine: 
- I've replaced the interposer for another one: same result. 
- I've replaced the D4 Vision Processor Board: same result.

Next, I've connected one of our RealSense assemblies (i.e. depth module + vision board + interposer) to an integrated RealSense D415 (https://www.intelrealsense.com/depth-camera-d415/) with the same Firmware as our assemblies. They were connected vi the trigger channel (pins 5 and 9). I've conducted the following tests using the RealSense Viewer GUI:

1. Set assembly to Master, set integrated device to Slave. Started slave. Result: immediate "Left IC2 Config Error"
2.  Set assembly to Master, set integrated device to Slave. First started Master, followed by slave. Result: sync achieved, and can be reliably reproduced
3. Switched Master and Slave and repeated test 2: everything OK and Slave stream can be reliably stopped and restarted
4. Stop Master stream. Quickly after "Left IC2 Config Error" is reported
5. Rest Slave device and repeated test 1 in (so now the assembly as Slave): "Left IC2 Config Error" again.

However, I could not reliably reproduce this behavior on our system. In one system this behavior was occurring for a whole morning, but somehow disappeared in the afternoon. To our understanding nothing had changed in the setup in the time in between. While the system was correctly working, the Master could also be stopped without affecting the Slave.

As such, in '2' Mode we sometimes see camera not functioning without a Master, while in other cases they do. This non-reproducable behavior is giving us headaches so I hope that you have some thoughts that could explain this.

MartyG-RealSense, 2023-01-03:
Are there any environmental temperature variations between morning and afternoon that could cause a malfunction to occur later in the day (i.e is the weather hotter in the morning than in the afternoon, potentially overheating the camera's internal electronics).

MartyG-RealSense, 2023-01-06:
Thanks very much @teunhoevenaars for the detailed update!

There is some additional sync cable design information provided by my Intel RealSense colleagues at the link below.  The information relates to the electronics schematic drawing for the cable in the hardware sync white-paper.

https://support.intelrealsense.com/hc/en-us/community/posts/360050894574/comments/360013337454

MartyG-RealSense, 2023-01-12:
Hi @teunhoevenaars  Do you require further assistance wih this case, please?  Thanks!

closed 2023-01-18 by MartyG-RealSense