realsenseai/librealsense #11248: YoloV7 and RealSense RuntimeError: No device connected error

NeroRGB, 2022-12-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { Arducam IMX219 } | 
| Operating System & Version |  Linux (Ubuntu 14/16/17) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson |
| Language                            |  {python}                          | 
| Segment			|  {Artificial Vision}                          | 

### Issue Description
I have installed YoloV7 and RealSense using this video as tutorial: [https://www.youtube.com/watch?v=o42BFhtVqyo&t=0s](https://www.youtube.com/watch?v=o42BFhtVqyo&t=0s) 
When I try to run the detect_RS.py file it shows me this message:

Traceback (most recent call last):
    File "detect_RS.py", line 179, in <module>
       detect()
    File "detect_RS.py", line 64, in detect
       profile = pipeline.start(config)
RuntimeError: No device connected 

I already installed NanoCamera and CSI-Camera, tried both and work fine, but this program can't seem to detect the camera.

MartyG-RealSense, 2022-12-21:
Hi @NeroRGB  The **JetsonHacks** installation procedure and subsequent pyrealsense2 installation procedure in the video look okay.

Is the camera able to be detected if you try the simple pyrealsense2 test script at the link beow, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

NeroRGB, 2022-12-22:
> Hi @NeroRGB The **JetsonHacks** installation procedure and subsequent pyrealsense2 installation procedure in the video look okay.
>
> Is the camera able to be detected if you try the simple [...]

Hi! I already tried that code and the camera is still undetected, same error

MartyG-RealSense, 2022-12-22:
The RealSense SDK and its pyrealsense2 Python wrapper is designed for Intel RealSense camera devices.  Your camera model is listed at the top of this case as **Arducam IMX219**.  A RealSense SDK script designed specifically for RealSense camera hardware would not be able to access a non-RealSense camera even if it was only the RGB image that was being accessed.

The YouTube video does not list the camera model but seems to assume that a RealSense camera is being used.

It is possible to import data into the SDK from non-RealSense cameras via the SDK's **software device** interface, but this would not be helpful for your YoloV7 project and software device works best on C++ as it has problems on Python.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

closed 2023-01-06 by MartyG-RealSense