realsenseai/librealsense #11249: SR305 : RGB and Depth misalignment on Windows

tanguy-magne, 2022-12-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | 6.27.3.0 | 
| Operating System & Version |   {Win 10 and Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  5.15.0-56-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2                        | 
| Language                            |  RealSense Viewer                          | 
| Segment			|  --                          | 

### Issue Description
I am using an SR305 camera in a fixed setup. The camera is on a tripod. I want to use it on both windows and an ubuntu machine.
However, the results produced are not the same (see images below).
[image]
This is the result I get on windows, which is not fine as the depth and the RGB are not aligned (see the white part in the background).

[image]
This is the result I get on ubuntu, which is fine, and I would like to have it on windows as well.

Do you have any idea why this happens ?
Thank you very much

MartyG-RealSense, 2022-12-22:
Hi @tanguy-magne  A key difference between the Windows and Ubuntu versions of the RealSense SDK is that the backend of the Windows version is based on **Microsoft Media Foundation** by default, whilst the Linux version's default backend is based on the **UVC** protocol.  

There is the option to build the SDK from source code on Windows with CMake instead of installing it with the Intel.RealSense.SDK-WIN10 installer file.  If the CMake build flag **FORCE_RSUSB_BACKEND** is set to True then the SDK should use a WinUSB rewrite of the UVC protocol as its backend instead of Media Foundation.

https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide

tanguy-magne, 2022-12-22:
Ok, I will probably try to build the SDK from source. Thanks!

The black square is due to a difference in depth. The white area is closer to the camera than the green background. Thus there is a gap. I expect it, and I am not bothered by it in the ubuntu version.

I do have a single SR305 camera in a fixed position, and I am swapping the other end of its USB cable between Windows and Linux computers (Linux for dev, but for the actual use, I would have windows).

Thank you for your clear answer. I will let you know if building the SDK from source make it work.

MartyG-RealSense, 2023-01-05:
Hi @tanguy-magne  Do you require further assistance with this case, please?  Thanks!

tanguy-magne, 2023-01-05:
Hi,
Sorry for the lack of answers. I haven't had the time to build the SDK from source, and will probably not have the time for it.
In the end, I used the Linux version.
I think you can close the issue. Thank you for your help :)

closed 2023-01-05 by MartyG-RealSense