realsenseai/librealsense #11251: Connecting to rs server on virtual machine from another windows machine over network

igor-aris, 2022-12-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435                                                       | 
| Firmware Version                   | 05.14.00.00| 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  2                          | 
| Language                            |                            | 
| Segment			|                          | 

### Issue Description
My camera is connected to a windows machine Server1. I'm trying to connect to the camera from another windows machine Server2 over network. For that, I cloned [this](https://github.com/IntelRealSense/librealsense) repo and generated a build with CMake 3.25.0. I used default flags and just enabled `BUILD_NETWORK_DEVICE`. After that I compiled the `realsense-viewer` with VS2022 and started it. When I select `Add network device` I get **`Conenct to a Linux system running rs-server`** window.

Then, I found [this](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-2-building-from-source) article that says the rs-server can run only on Linux.

What are my options? 

1. Can I run a virtual machine with linux on Server1 and run the rs-server from within that virtual machine? Will rs-server detect the camera? Will I then be able to connect to that camera from Server2?
2. Are there other out-of-box alternatives to rs-server that I can use?

[image]

MartyG-RealSense, 2022-12-24:
Hi @igor-aris  If you are able to use Python on your Windows machines, the predecessor of the rs-server system called **EtherSense** provides RealSense ethernet networking.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

You could also explore the possibility of simulating Linux on one of your Windows machines with **Windows Subsystem for Linux 2 (WSL2)**

https://learn.microsoft.com/en-us/windows/wsl/install

WSL2 tends to be more problematic to use with RealSense cameras than a full Linux installation though (such as Windows and Linux both installed on a dual-boot PC).  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/10506#issuecomment-1125860744 had camera detection problems when using WSL2. They managed to detect the camera with Windows 11 after adjusting some settings, but did not succeed with Windows 10.

If you would like to use a Linux virtual machine on a Windows PC, googling for the term **virtual machine windows linux** provides a number of guides that indicate that this is possible.  The VM tool that is selected though should have USB controller emulation in order for it to be used with RealSense cameras.  Intel recommend **VMWare Workstation Player**, though other VM tools that have USB emulation have been successfully used with RealSense cameras.

Another possibility would be to remotely control one Windows PC from the other Windows PC.  An example of such a tool is Google's **Chrome Remote Desktop** at https://remotedesktop.google.com/

igor-aris, 2022-12-24:
So, if I choose the VM option, I can run `RealSense Viewer` on the Windows host and run the `rs-server` from the Linux VM or should I run both `RealSense Viewer` and `rs-server` from the Linux VM?

MartyG-RealSense, 2022-12-25:
Even though EtherSense uses Python, it appears to be specifically designed for Linux and its directory structure and so would be unsuitable for use with two Windows PCs.  My apologies.

In an rs-server ethernet networking system, RealSense Viewer will be on the host (the Windows computer) and rs-server will be on the Linux computer.

With a remote access tool such as Chrome Remote Desktop, you are seeing the content of one computer on the other computer's screen but whatever action you take (running programs, making changes, etc) happens on the computer that is being remotely controlled.

closed 2023-01-06 by MartyG-RealSense