realsenseai/librealsense #11256: Cannot connect camera L515

liyuan1a, 2022-12-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (01.05.08.01) | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.51.1}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {others }                          | 

### Issue Description
We can't connect all the 3 camera L515s, even if the Intel RealSence Viewer monitors three L515 devices, but we can't add it from the "Add Source" Button.
It is worth noting that we were able to connect all the devices normally a few days ago. 
During this period, we just shut down the computer, and did not change or process the connection line.
[image]

MartyG-RealSense, 2022-12-25:
Hi @liyuan1a  Please first check that permission to access the camera and permission for applications to access the camera are both set to On in the Windows camera privacy settings.  This can be performed using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

liyuan1a, 2022-12-26:
We have checked that permission to access the camera and permission for applications to access the camera are both set to On in the Windows camera privacy settings. But it still has such problems.

MartyG-RealSense, 2022-12-26:
It looks as though you are able to list at least one of the L515 cameras as it is automatically detected and listed when the Viewer is launched.  But other L515 cameras are not accessible through the Add Source button.

I tested with two L515 cameras in the standalone version of the 2.51.1 RealSense Viewer and the second L515 was listed in the Add Source option.

[image]

Can you try the cameras in the standalone **Intel.RealSense.Viewer.exe** version of the Viewer if you have not done so already, please?  This version can be downloaded from the 'Assets' file list of the 2.51.1 SDK version.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

[image]

The standalone Viewer should be launched with a double-leftclick from the location on your computer that you downloaded the .exe to and not with the launch shortcut on the Windows desktop.

MartyG-RealSense, 2023-01-02:
Hi @liyuan1a  Do you require further assistance with this case, please?  Thanks!

liyuan1a, 2023-01-03:
I've tried to use the **Intel.RealSense.Viewer.exe** program several times to start, but it didn't work. I tried to turn off the computer, cut off the power, and then restart it. After several attempts, the software can identify all L515 cameras.

MartyG-RealSense, 2023-01-03:
Let's try uninstalling and re-installing the drivers of the L515 cameras from the Windows **Device Manager** to eliminate the possibility that the installed drivers are damaged.

1.   Unplug all three L515 cameras from the computer. 

2. Find and open the Device Manager interface by typing **device** into the text box at the bottom corner of your Windows screen.

[image]

3.  In the Device Manager window, go to the **View** menu option and select **Show hidden devices** to list the Depth and RGB driver pairs for each of the RealSense cameras that have previously been attached to the computer.  So there should be three pairs of L515 Depth and RGB drivers on your particular computer for your three individual L515 cameras.  

[image]

[image]

4.  Right-click on a Depth or RGB driver and select the **Uninstall device** menu option.  Repeat until all three of the pairs of Depth and RGB drivers have been uninstalled.

[image]

5.  Plug the L515 cameras into the computers.  The drivers should automatically reinstall into the Device Manager.  Then open the RealSense Viewer to see whether their ability to be detected consistently has improved.

liyuan1a, 2023-01-06:
I tested it according to your suggestion. After several tests, it didn't help. At present, the software **Intel.RealSense.Viewer** can only detect all L515 cameras by restarting the computer (turning off the power).

MartyG-RealSense, 2023-01-06:
If all cameras are identified only after shutting down the computer and then re-starting it with the power button (a **cold boot** and not a reset), the problem may be related to the start-up **BIOS** of the particular PC that the cameras are being used on and not with the camera hardware.  

The RealSense team did a lot of testing of this problem in previous cases from a few years ago. It was thought that the problem may be related to the USB cable or USB port / controller.  If multiple cameras are affected then it is more likely to be related to the USB port / USB controller on the computer instead of the USB cable.

If the cameras are attached to a USB hub then changing the hub may help, as a different hub would likely use a different brand of USB controller.  A hub that is powered by the mains electricity socket instead of attached to the computer will also be more stable than a hub that is attached to the computer and draws its power from it, especially when multiple cameras are being used simultaneously.

MartyG-RealSense, 2023-01-12:
Hi @liyuan1a  Do you have an update about this case that you can provide, please?  Thanks!

liyuan1a, 2023-01-18:
Because my problem occurred suddenly after using it for a period of time, it is difficult to judge whether it is caused by the hub. At present, the software **Intel.RealSense.Viewer** can only detect all L515 cameras by restarting the computer (turning off the power).

MartyG-RealSense, 2023-01-18:
As mentioned above at https://github.com/IntelRealSense/librealsense/issues/11256#issuecomment-1373510141 it sounds like a BIOS issue on the particular computer that you are using.  The BIOS is stored in a chip on the computer and controls the computer's initial start-up configuration before Windows has loaded when the power button is pressed to boot it up from the shut-off state.  I note though that the cameras were able to be previously detected after boot until recently.

In cases on Windows computers where there is an "it worked until yesterday" type situation, the installation of a Windows update may be the cause of the problem.  If you typr **view your update history** into the text box at the bottom of the Windows screen (next to the corner button where the computer is shut down) then you can see whether there were Windows updates that coincide when the problem started 3 weeks ago at the time of writing this.

liyuan1a, 2023-01-19:
Thank you for your help. In general, this has not had a serious impact on the use of L515 cameras. thank you.

MartyG-RealSense, 2023-01-19:
You are very welcome.  I'm pleased that I could be of assistance.  :)

closed 2023-01-24 by MartyG-RealSense