realsenseai/librealsense #11258: Cannot Detect IMU D435i

dariandzirko, 2022-12-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 0.5.14.00.00 | 
| Operating System & Version | Linux (Ubuntu 22) | 
| Kernel Version (Linux Only)    |  5.15.0-56-generic  | 
| Platform                                 | PC  |
| SDK Version                            | 2.53.1 | 
| Language                            | { python } | 

### Issue Description
After following the tutorial for the linux installation and upgrading the firmware on my D435i, I can successfully viewer the RGB and Depth information in RealSense Viewer. I cannot view and IMU information and cannot even find the rs-imu-calibration script after 2 installs of the packages. Even trying the rs-motion example leads to a response of Device supporting IMU not found, which should not be the case. 

I have not tried building it from source. 
Tutorial I followed to install https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

MartyG-RealSense, 2022-12-26:
Hi @dariandzirko  Is the camera name shown as D435i at the top of the RealSense Viewer's options side-panel?

[image]

dariandzirko, 2022-12-26:
Oh wow the order and box said D435i but RealSense Viewer says D435. I missed that thanks.

MartyG-RealSense, 2022-12-26:
The RealSense SDK identifies the camera name from the camera components that the firmware driver can detect.  So it may indeed be a D435i unit but is being identified by the firmware as an IMU-less D435 because the firmware cannot detect the IMU component for some reason.

closed 2023-01-09 by MartyG-RealSense