realsenseai/librealsense #11260: Ethernet configuration

IndiGleb, 2022-12-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |                            { D400 }                       | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |                     PC/Raspberry Pi                 |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |                             {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Hello. I found disagreement in documentation "realsense ethernet". In documentaion valid configuration : 
[image]
But in realsense viewer i can't found this config:
[image]
[image]
[image]
Please help me with that. I need this configuration

MartyG-RealSense, 2022-12-26:
This issue is a continuation of https://github.com/IntelRealSense/librealsense/issues/11234

Hi @IndiGleb  I will highlight to my Intel RealSense colleagues your issue with the lack of a minimum FPS on your D455 camera. when using rs-server ethernet networking.

IndiGleb, 2022-12-27:
[image]
maybe I don't understand something, but the configuration is in rs-server

IndiGleb, 2022-12-27:
In RsServer.cpp

IndiGleb, 2022-12-27:
[image]
[image]
This configuration too empty

MartyG-RealSense, 2022-12-27:
Resolutions 848x480 and 640x360 are not listed as supported for a Network Camera in the chart of available resolution / FPS combinations. 

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#36-software-limitations

IndiGleb, 2022-12-27:
But resolution 640x360 working on client 
[image]

IndiGleb, 2022-12-27:
And i can stream this

MartyG-RealSense, 2022-12-27:
Thanks very much for the information about the network stream modes that are currently accessible in the Viewer.

Can you describe what you mean by it being 'empty' please.

IndiGleb, 2022-12-27:
in realsense viewer i can't found this config

MartyG-RealSense, 2022-12-27:
The RealSense Viewer on the client would not be a network camera.  It is a network camera when the camera on the client computer is accessed from the Viewer on the host computer.

IndiGleb, 2022-12-27:
https://github.com/IntelRealSense/librealsense/issues/11260#issuecomment-1365035933
When can I get an answer ?

MartyG-RealSense, 2022-12-27:
I will pass you information about your question as soon as I receive it from my colleagues.  It will likely be this week but not today.

IndiGleb, 2022-12-27:
In SDK 2.39.0 has been changed configuration. In documentation used 2.32. I think documentation outdated

IndiGleb, 2022-12-27:
In this documentation 
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

MartyG-RealSense, 2022-12-27:
At the time that the guide was introduced, 2.34.0 was the minimum.

MartyG-RealSense, 2022-12-28:
Hi @IndiGleb  I have received information from my Intel RealSense colleagues regarding your question about the absence of 5 FPS on D455.

***************

The network camera feature is just a proof of concept (POC), not an official RealSense product.  When the ethernet white-paper document was published, the D455 camera model was not yet released and Intel decided to support 5 FPS on D455 instead of 6 FPS.

**********

With experimental POC features in RealSense white papers (which also include External Synchronization genlock hardware sync and depth compression by colorization), RealSense users are free to attempt using them in projects but they are not validated as mature features and may not always function as decribed in their paper, especially for camera models introduced after the paper was published.

IndiGleb, 2022-12-29:
But configuration with 5 fps not supported

IndiGleb, 2022-12-29:
If i use realsense on ethernet

MartyG-RealSense, 2022-12-29:
You are correct, 5 FPS on D455 is not supported because the D455 model was released after the rs-server ethernet tool was developed.

IndiGleb, 2022-12-29:
and 6 fps ?

MartyG-RealSense, 2022-12-29:
Yes, 6 FPS should work on 400 Series cameras that support that speed, such as D415 and D435 / D435i.

IndiGleb, 2022-12-29:
will this config be added?

MartyG-RealSense, 2022-12-29:
There are no plans to add 5 FPS support for D455 unfortunately as the rs-server ethernet networking system is a 'proof of concept' rather than a tool that is provided with ongoing updates.

IndiGleb, 2022-12-29:
anyway thanks you for help

closed 2023-01-04 by MartyG-RealSense