realsenseai/librealsense #11263: No Video Capture Device Error rs2_create_device

yeric1789, 2022-12-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 22.04 LTS)  | 
| Kernel Version (Linux Only)    |  (5.15.0-56-generic)                                         | 
| Platform                                 | PC (x86 CPU)  |
| SDK Version                            |  { 2.53.1}                          | 
| Language                            |  {C/C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
I have been attempting to install on a new Ubuntu distro 22.04 LTS, however when running my program which previously worked on Ubuntu 20.04 I find the following error being thrown:
```
RealSense error calling rs2_create_device(info_list:0x555e5b7a8c00, index:1):
    /dev/video0 is no video capture device Last Error: Invalid argument
```

When I go to list my devices out for my computer I see the following and indeed the Depth camera does not correspond with /dev/video0

```
$: v4l2-ctl --list-devices
ipu3-imgu (PCI:0000:00:05.0):
	/dev/video0
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8
	/dev/video10
	/dev/video12
	/dev/video13
	/dev/media1

Intel IPU3 CIO2 (PCI:0000:00:14.3):
	/dev/video1
	/dev/video2
	/dev/video9
	/dev/video11
	/dev/media0

Intel(R) RealSense(TM) Depth Ca (usb-0000:00:14.0-1):
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video17
	/dev/video18
	/dev/video19
	/dev/media2
	/dev/media3
```
I have tested out the realsense-viewer and recorder which seems to work fine. I only have one device connected. I have also checked my code and found that the problem starts when I query for the devices. specifically looping through any connected devices:
```
for (rs2::device &&dev : ctx.query_devices()){
...
```

 I have seen a similar issue from here  #10072. I have checked for memory leaks which don't seem to present. I have seen that they used docker (which currently I am not using). Is there any way to get this fixed short of a custom build? Thanks.

MartyG-RealSense, 2022-12-27:
Hi @yeric1789  If you are only using one camera, an alternative to looping thorugh all connected devices may be to query for the first depth sensor that can be detected.  For example in C++:

**auto sensor = selection.get_device().first<rs2::depth_sensor>();**

yeric1789, 2022-12-27:
OK thanks that was able to fix it.

closed 2022-12-27 by yeric1789

MartyG-RealSense, 2022-12-27:
That's great to hear, @yeric1789 - thanks very much for the update!