realsenseai/librealsense #11264: 3D Visualization in c# how to do that?

noodle875, 2022-12-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  { C# }                          | 

Hi, I'm developing a realsense d435i in C#.
I've been looking for resources related to realsense 3D Visualizaition in C# for a day and haven't found anything.
I would like to find the material by posting here.

MartyG-RealSense, 2022-12-27:
Hi @noodle875  If you wish to generate a 3D pointcloud then the RealSense SDK's 'C# Cookbook' page of example code has a script called **Generics Support** that does this.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#generics-support

noodle875, 2022-12-27:
Yes, I saw its contents. But I don't know how I can use it for 3D visualization.
If you know, can you upload the C# code?

MartyG-RealSense, 2022-12-27:
Can you describe what you mean by '3D visualization' please.  For example, do you want to convert a pointcloud to a solid 3D model?

noodle875, 2022-12-27:
I haven't learned image processing, so I don't know the details, but anything is fine.
The 3D visualization provided by the Depth Quality Tool is also good.

MartyG-RealSense, 2022-12-27:
The default '3D' mode of the Depth Quality Tool is a pointcloud like the one produced by the Generics Support C# script.

[image]

noodle875, 2022-12-27:
Oh I see. thanks.
I didn't know Depth Quality Tool was made in C#

But I need C# code not Depth Quality Tool.
Could you upload some sample source code in C# related to that from data acquisition to data visualization ?

MartyG-RealSense, 2022-12-27:
The Depth Quality Tool is C++ but pointclouds can be generated with different programming languages (C++, C# and Python).

Generics Support is the best available pointcloud script for C#.

closed 2022-12-28 by noodle875