realsenseai/librealsense #11269: Can´t get 300 fps -- Couldn't resolve requests -- D400

Calvincs77, 2022-12-27:
Hello i am trying to configure the d455 and d405 camera for high speed capture mode using python

i have verified that my camera is detected as USB 3 as mentioned in  #2818 Couldn't resolve requests 

[image]

And also verifying the firmware version higher than 5.12.4.0. from [https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435#section-2-2-enabling-infrared-stream-for-monochrome-image-capture](url)

[image]

I have created two codes one with 848x480 - 90 fps and it works perfectly but the code with 848 x 100 - 300 fps Couldn't resolve requests.

Here´s my code 

```
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

 #####works well with config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 90)#####
config.enable_stream(rs.stream.depth, 848, 100, rs.format.z16, 300) 
pipeline.start(config)

try:
    while True:

        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        if not depth_frame :
            continue

        depth_image = np.asanyarray(depth_frame.get_data())
        #Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', depth_colormap)
        cv2.waitKey(1)

finally:
    pipeline.stop()
```

Get error  File "c:\Users\57310\Documents\Untitled-1.py", line 16, in <module>
    pipeline.start(config)
RuntimeError: Couldn't resolve requests

Thanks for your help

MartyG-RealSense, 2022-12-28:
Hi @Calvincs77  I tested selecting 848x100 resolution on D455 in the RealSense Viewer.  The fastest supported FPS for that resoution was **100** FPS.  On the D435 camera model though, 300 FPS was available.

[image]

In regard to D405, the absence of 300 FPS mode (and the non-availability of 848x100 resolution) was recently reported by another RealSense user at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/11951486064019-Changing-Fov-by-cropping-images-on-D405-depth-camera

[image]

I will raise the issue of 300 FPS support for D405 and D455 with my Intel RealSense colleagues.  Thanks for the report!

MartyG-RealSense, 2022-12-28:
Hi @Calvincs77  After discussion with my colleagues, official Intel feature requests to add 300 FPS support on D455 and the 848x100 / 300 FPS resolution on D405 have been created.

This issue should be kept open whilst these change requests are active, but you do not need to take any further action.  Thanks again!

Calvincs77, 2022-12-28:
Thank you for your quick response, we will be waiting with our team for these requests.

Regards

MartyG-RealSense, 2025-05-21:
It was decided that 300 FPS support for D405 and D455 would not be implemented.  For D405 in particular it was because the feature would not work due to that model's lack of a laser component.  For D455 it was not a development priority.

closed 2025-05-21 by MartyG-RealSense