realsenseai/librealsense #11276: Hello realsense example crashes on Jetson Jetpack 4.2.2 using Intel D435

cesarpgouveia, 2022-12-29:
----------------------------------------------------------------------------------------------------

| Environment                           |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                   | 
| Operating System & Version |   Linux (Ubuntu 18.04)  |                                      
| Platform                                 | NVIDIA Jetson Jetpack 4.2.2  |
| SDK Version                            |  2.53.1                           | 
| Language                            |  C++                           |  

### Issue Description

Hey guys,

I'm having problems running the hello-realsense example using NVIDIA Jetson Jetpack 4.2.2 and an Intel D435.

I built the most recent version of your code (release 2.53.1) with the following flags:

 cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true

Then I run the hello-realsense example, the example was running smoothly. But then, after a few hours of running non-stop I got the following error:

RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5596bafed0):
    Device disconnected. Failed to reconnect: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error15000

And after I'm not able to run the example or anything related to the D435:

RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x55841fe630):
    Frame didn't arrive within 15000

Is there any incompatibility with the NVIDIA Jetpack version that I'm using (Jetpack 4.2.2)? I have some things in mind to try, like updating the camera firmware, but if you have any direction to point me or if this is a known issue, I would appreciate it.

Thanks,
César.

MartyG-RealSense, 2022-12-29:
Hi @cesarpgouveia  The Jetson instructions for installing the SDK from source code state that the procedure has been tested with JetPack 4.2.3, so 4.2.2. is likely okay and the fact that you were able to successfully run a program for a few hours non-stop would seem to confirm that the SDK build works correctly.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

If you are using SDK 2.53.1 then it is recommended to update the camera firmware driver to version **5.14.0.0**.  Again though, the rs-hello-realsense project was able to run correctly for a prolonged period of time, suggesting that the problem is not with the firmware.

If you would like to update your JetPack, **4.6** would be a safe choice for using the SDK with a Jetson running Ubuntu 18.04.

Are you able to run programs again if you shut down the Jetson board (not reset it) and then reboot it from the turned-off state?

In regard to the error, a **xioctl(VIDIOC_S_FMT)** error indicates that the problem occurred because a **kernel** conflict occurred.  If kernel related problems occur then the SDK can be built with the **RSUSB method**, which bypasses the kernel and makes the build not dependent on Linux versions or kernel versions and not requiring kernel patching.  An RSUSB build can be performed by adding **-DFORCE_RSUSB_BACKEND=true** to the CMake build instruction.

`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DFORCE_RSUSB_BACKEND=true`

cesarpgouveia, 2022-12-30:
First of all, thanks for all the suggestions!

So, I tried to build with -DFORCE_RSUSB_BACKEND=true and now I don't get the following error:
Device disconnected. Failed to reconnect: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error15000

However, the camera still freezes after some hours of running the hello-realsense example non-stop with the following error: 
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x559b9d35c0):
    Frame didn't arrive within 15000

After that, as it happened before, the camera is unavailable until I restart the board:
RealSense error calling rs2_pipeline_start(pipe:0x5589ea1520):
    No device connected

I just checked my firmware version, and in fact, it is a little outdated compared to the one you suggested:
Connected devices:
1) Name: Intel RealSense D435, serial number: 948122070895, update serial number: 948123023010, firmware version: 05.11.01.100, USB type: 3.2

So my next step will be to update that and I will get back to you and to this thread.

Thanks,
César.

MartyG-RealSense, 2022-12-30:
Thanks very much for the update.  Firmware 5.11.1.100 is a very old firmware that dates back to February 2019 and should definitely be updated to a modern version.

cesarpgouveia, 2023-01-02:
Hey @MartyG-RealSense,

So I updated the firmware version to the one you suggested and now the hello realsense example runs without issues for several days (at least it has been running non-stop since 31/12/2022 and still hasn't crashed). I'm going to do some more stability tests but I think this thread can be closed because I think the issue here was the firmware being outdated.

Thank you so much for your help, and have a wonderful year!
César.

MartyG-RealSense, 2023-01-02:
You are very welcome, @cesarpgouveia - it's excellent to hear that a firmware update was successful for you,  Thanks very much, and I wish a great year for you too!  :)

closed 2023-01-03 by cesarpgouveia