realsenseai/librealsense #11283: Unsupported Linux Kernel Version 5.15

samchiRobot, 2023-01-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-56-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.53.1                         | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
Hello, I want to install Realsense SDK on my Desktop.
When I execute this command, 
``` bash:command
./scripts/patch-realsense-ubuntu-lts.sh
```
Install is failed with below message.
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.07.1-2build1).
build-essential is already the newest version (12.8ubuntu1.1).
git is already the newest version (1:2.25.1-1ubuntu3.6).
linux-headers-generic is already the newest version (5.4.0.135.133).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Unsupported kernel version 5.15.0-56-generic . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only
```

So, I wonder when LibRealsense will support on Linux kernel version 5.15.
Thanks in advance!

MartyG-RealSense, 2023-01-02:
Hi @samchiRobot  Support for kernels 5.13 and 5.15 on Focal and 5.15 on Jammy was added in RealSense SDK version **2.53.1**.  They have their own kernel patch script called **./scripts/patch-realsense-ubuntu-lts-hwe.sh**

closed 2023-01-18 by MartyG-RealSense

MartyG-RealSense, 2023-09-21:
Hi @dllevat-airvant  Kernel 5.15 can be used with Ubuntu 20.04 but support in the patch script for doing so was not added until librealsense 2.53.1.

[image]