realsenseai/librealsense #11292: Unable to run more than two pipelines (Segmentation Fault)

maciejandrzejewski-digica, 2023-01-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Debian 11 (Raspbian RPI4)  | 
| Kernel Version (Linux Only)    |  5.15.56-v8+                                        | 
| Platform                                 | Raspberry Pi 4B  |
| SDK Version                            |  v2.51.1                          | 
| Language                            |  C/C++                          | 
| Segment			|  AI/CV                          | 

## Issue Description

I'm running single pipeline for RGB and Depth data just fine.

In order to acquire accelerometer and gyroscope data I have created additional two pipelines which I have planned to run in two separate threads. I would like to run the sensors with full speed and they are running with different frequencies that is why I figured out to run them in separate pipelines.
For a test I have created the accl & gyro pipelines just to check the data is ok but I have found out that running more then two triggers Segmentation Fault error. If I run only RGB-Depth and accl or gyro it is working fine.

This code works fine:
```C++
int RealSenseVideoSource::open(const char *path)
{
    int ret{};
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH);
    cfg.enable_stream(RS2_STREAM_COLOR);
    if(m_sensInertialEnabled)
    {
        rs2::config cfgAcc;
        cfgAcc.enable_stream(RS2_STREAM_ACCEL);
        auto ppAccel = m_accel.start(cfgAcc);
        if(! ppAccel)
        {
            msgError() << "Cannot open accelerometer pipeline" << endl;
            ret |= -2;
        }

//        rs2::config cfgGyro;
//        cfgGyro.enable_stream(RS2_STREAM_GYRO);
//        auto ppGyro = m_gyro.start(cfgGyro);
//        if(! ppGyro)
//        {
//            msgError() << "Cannot open gyroscope pipeline" << endl;
//            ret |= -3;
//        }
    }
    auto pp = m_cam.start(cfg);
    if(!pp)
    {
        m_isOpened = false;
        msgError() << "Cannot open video capture" << endl;
        ret |= -1;
        return ret;
    }

    if(msgIsDebug())
        m_rprint = std::make_unique<rs2::rates_printer>();

    m_isOpened = true;

    return ret;
}
```

This one works as well:
```C++
int RealSenseVideoSource::open(const char *path)
{
    int ret{};
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH);
    cfg.enable_stream(RS2_STREAM_COLOR);
    if(m_sensInertialEnabled)
    {
//        rs2::config cfgAcc;
//        cfgAcc.enable_stream(RS2_STREAM_ACCEL);
//        auto ppAccel = m_accel.start(cfgAcc);
//        if(! ppAccel)
//        {
//            msgError() << "Cannot open accelerometer pipeline" << endl;
//            ret |= -2;
//        }

        rs2::config cfgGyro;
        cfgGyro.enable_stream(RS2_STREAM_GYRO);
        auto ppGyro = m_gyro.start(cfgGyro);
        if(! ppGyro)
        {
            msgError() << "Cannot open gyroscope pipeline" << endl;
            ret |= -3;
        }
    }
    auto pp = m_cam.start(cfg);
    if(!pp)
    {
        m_isOpened = false;
        msgError() << "Cannot open video capture" << endl;
        ret |= -1;
        return ret;
    }

    if(msgIsDebug())
        m_rprint = std::make_unique<rs2::rates_printer>();

    m_isOpened = true;

    return ret;
}
```

But running both of the inertial sensors pipeline (plus the RGB-Depth pipeline) will trigger Segmentation Fault at the marked line:

```C++
int RealSenseVideoSource::open(const char *path)
{
    int ret{};
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH);
    cfg.enable_stream(RS2_STREAM_COLOR);
    if(m_sensInertialEnabled)
    {
        rs2::config cfgAcc;
        cfgAcc.enable_stream(RS2_STREAM_ACCEL);
        auto ppAccel = m_accel.start(cfgAcc);
        if(! ppAccel)
        {
            msgError() << "Cannot open accelerometer pipeline" << endl;
            ret |= -2;
        }

        rs2::config cfgGyro;
        cfgGyro.enable_stream(RS2_STREAM_GYRO);
        auto ppGyro = m_gyro.start(cfgGyro);  // <------------------------------------- Segmentation Fault
        if(! ppGyro)
        {
            msgError() << "Cannot open gyroscope pipeline" << endl;
            ret |= -3;
        }
    }
    auto pp = m_cam.start(cfg);
    if(!pp)
    {
        m_isOpened = false;
        msgError() << "Cannot open video capture" << endl;
        ret |= -1;
        return ret;
    }

    if(msgIsDebug())
        m_rprint = std::make_unique<rs2::rates_printer>();

    m_isOpened = true;

    return ret;
}
```

When I step-in to rs2::pipeline::start() it faults at `rs2_pipeline_start_with_config()`.

How many pipelines can be run simultaneously?

Does above approach is the right one? I need one pipeline with RGB & Depth synchronized and full speed streams of accelerometer and gyroscope data.

MartyG-RealSense, 2023-01-04:
Hi @maciejandrzejewski-digica  Defining seperate pipelines for depth / color and IMU manually for a single camera (not multiple cameras) is a method that works best on Python.  For C++, using **callbacks** instead is a better solution for streaming depth, color and IMU simultaneously.  This is discussed at https://github.com/IntelRealSense/librealsense/issues/6404

The RealSense SDK's C++ example program **rs-callback** also demonstrates efficient simultaneous use of depth, color and IMU.

https://github.com/IntelRealSense/librealsense/tree/master/examples/callback

If you prefer to use your current method in the scripting above, I would recommend putting gyro and accel on the same pipeline instead of dividing them into two individual pipelines.

MartyG-RealSense, 2023-01-11:
Hi @maciejandrzejewski-digica  Do you have an update about this case that you can provide, please?  Thanks!

maciejandrzejewski-digica, 2023-01-16:
Hi @MartyG-RealSense 
Sorry for late response. I have refactored the code to use the callback and single pipeline and it is working. Later on I have found that you suggested the same.
Anyway this is a bug in the library (Segmentation Fault) when max pipelines reached.
For me it can be closed now.
Thanks for help!

MartyG-RealSense, 2023-01-17:
Thanks very much @maciejandrzejewski-digica for the update - it's great to hear that you achieved a working solution.  As you suggest, I will close the issue.  Thanks again!

closed 2023-01-17 by MartyG-RealSense