realsenseai/librealsense #11296: X Error of failed request:  BadWindow (invalid Window parameter)

arthurgassner, 2023-01-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  05.13.00.50                                      | 
| Platform                                 | PC |
| SDK Version                            |    2.45.0.3278 
| Language                            |  -                          | 
| Segment			|  -                         | 

### Issue Description

I am using Intel RealSense Viewer v2.53.1 to work with Intel cameras (D400).

I start the viewer by running `realsense-viewer` in a terminal.

When using the viewer, it randomly crashes (every ~3 clicks on the GUI), throwing 

```bash
X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  25 (X_SendEvent)
  Resource id in failed request:  0xa00009
  Serial number of failed request:  561186
  Current serial number in output stream:  561186
```

and forcing me to restart the viewer.

As anyone encountered this issue? Does someone know how to solve it?

Have a great year!

MartyG-RealSense, 2023-01-05:
Hi @arthurgassner  If you are using SDK version 2.53.1 then I would first recommend updating your firmware driver version from 5.13.0.50 to **5.14.0.0**.

In regard to this particular error, it has not previously been reported in relation to RealSense cameras.  I note though that you are using kernel **5.14.0-1055-oem**.  Kernel 5.14 is not supported by the RealSense SDK.  Kernels **5.13** and **5.15** are supported for Ubuntu 20.04.  

Using an unsupported kernel can result in unpredictable consequences in regards to stability.  If you are unable to change the kernel version then it is possible to bypass the kernel by building the SDK from source code with CMake using the **RSUSB backend** installation method.  An RSUSB build of the SDK is not dependent on Linux versions or kernel versions and does not require kernel patching.  It is best suited for use with a single camera however, whilst a kernel-patched build is ideal for multiple camera applications.

If updating your camera firmware driver to 5.14.0.0. did not resolve the error and you would like to attempt an RSUSB source code installation then instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

MartyG-RealSense, 2023-01-12:
Hi @arthurgassner  Do you require further assistance with this case, please?  Thanks!

arthurgassner, 2023-01-17:
Hi @MartyG-RealSense :)

I updated my Ubuntu version to 22.04, which updated my kernel to `5.15.0-58-generic`  and the issue was fixed.

Thank you so much for your quick reply

closed 2023-01-17 by arthurgassner

MartyG-RealSense, 2023-01-17:
You are very welcome, @arthurgassner - I'm pleased to hear that you achieved a solution.  Thanks very much for the update!